Atmosphere-NX/Atmosphere #1561: DNS.mitm disable fatal error

ttom832, 2021-07-10:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

N/A

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br> 
Trying to disable dns.mitm in order to allow save game transfer to cfw. 

[ Did you make any changes related to Atmosphère itself? ]</br>
No

[ If so, make sure to include details relating to what exactly you changed. ]</br>
N/A

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
Insert atmosphere!enable_dns_mitm = u8!0x0
Into [atmosphere] section of system_settings.ini
Try to boot atmosphere from Hekate primary. 
screen displays fatal error message and continuously tries to reboot but does not produce crash report.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

12.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 
  - [ Official release version 5.5.8 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] N/A
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-07-10:
`atmosphere!enable_dns_mitm = u8!0x0` is invalid format for settings, this will cause parse error.

Re-read documentation, you just need `enable_dns_mitm = u8!0x0` in the [atmosphere] section.

closed 2021-07-10 by SciresM

ttom832, 2021-07-10:
Thank you!
Regards- Sleep deprived