Atmosphere-NX/Atmosphere #1567: A fatal error occure when running Atmosphere. Program ID: 524850534f53454d

fuyukiakihiro, 2021-07-16:
### What's the issue you encountered?

I'm on 12.1.0, with atmos at 0.19.5, I updated fusee-primary.bin ofc. I got this message when I inject fusee-primary.bin:
"A fatal error occure when running Atmosphere.
Program ID: 524850534f53454d
Error Desc: Kernel Panic (0xf00)
Report saved to /atmosphere/fatal_errors/report_0000000001def6bd.bin"

### How can the issue be reproduced?

My switch won't boot after I update it to 12.1.0 from OFW via OTA, can't boot CFW too. So I got no choice, but downgrade it manually from PC using choidujour to 5.1.0. From there, I use atmosphere 0.8.5 to boot CFW, then update it to 12.1.0 using choidujourNX. Then, I update atmosphere to 0.19.5, insert it to my switch, then tried to inject the latest fusee too, and I got that error message. Until now, I can't boot to the latest atmosphere.

### System Firmware Version

12.1.0

### Environment?

- Atmosphere 0.19.5 with its fusee-primary.bin

### Additional context?

N/A

SciresM, 2021-07-16:
...upload the report file?

fuyukiakihiro, 2021-07-16:
here it is.

[report_0000000001def6bd.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6830270/report_0000000001def6bd.zip)

SciresM, 2021-07-16:
This issue is the same as that other guy had, a long time ago, where it's a panic when loading the initial process binary.

Please contact me on discord so I can look at your NAND, this panic should only occur when the INI is invalid which shouldn't be possible except with corrupt package2.

Of course, I note: `update it to 12.1.0 using choidujourNX`, so I'm not actually surprised by corrupt package2.

If you just want a fix and not to resolve the underlying problem, you could just flash good package2 manually to emmc using an unbrick guide.

SciresM, 2021-07-16:
My discord handle is SciresM#0524.

fuyukiakihiro, 2021-07-17:
I can't find you on discord with that. Is anything missing?

closed 2021-07-21 by SciresM

UCanthandlemeh, 2021-11-27:
> 

did you fix it its happening to me right now

UCanthandlemeh, 2021-11-27:
> My discord handle is SciresM#524.

can you help me out cant find you on discord

masagrator, 2021-11-27:
> > My discord handle is SciresM#524.
> 
> can you help me out cant find you on discord

SciresM#0524

UCanthandlemeh, 2021-11-27:
> > > My discord handle is SciresM#524.
> > 
> > 
> > can you help me out cant find you on discord
> 
> SciresM#**0**524

Thanks

JrRamos1524, 2022-05-11:
What was the solution? I have the same problem, and I don’t know what to do.

SciresM, 2022-05-11:
@JrRamos1524 make a new issue with a filled out template and upload the report file.

DRMon8, 2022-08-06:
Doesn't the problem fix? I also have same problem 😔

Masamune3210, 2022-08-06:
Dont do that, make a new issue and fill out the report please

DRMon8, 2022-08-16:
does this problem resolve?  at same problem too