Atmosphere-NX/Atmosphere #1570: Panic Occurred While Running Atmosphere after a black screen in Atmosphere

peter765, 2021-07-17:
## Bug Report

### What's the issue you encountered?

Was playing a game off a cartridge in Atmosphere, screen blacks out, booting into Switch OS on Sysnand, Emunand, or Stock always returns a black screen. 

### How can the issue be reproduced?

- Inject Fusee payload via Hekate 5.5.8
- Atmosphere splash shows up
- (screen turns black)
- Try to inject another payload
- Get the error shown in the attached screenshot

### Crash Report

Nothing in Crash Reports, however there is a file in the "fatal_errors" directory that I have attached

### System Firmware Version

12.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official atmosphere 0.19.5
- Do you have additional kips or sysmodules you're loading: None. 
- Homebrew software installed: None
- EmuMMC or SysNAND:
  -Issue present on both SysNAND and File-based EmuNAND

### Additional context?

Tried restoring nand backup. 
Console no longer boots with or without an SD card. 

fatal error and screenshot of error:
[issue docs.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6835326/issue.docs.zip)

SciresM, 2021-07-18:
So you're getting a kernel panic due to data abort in initial process creation -- during BLZ uncompress, in particular.

The address in FAR_EL1 looks like it has bit corruption in it, to me.

I would normally suggest checking for SD card corruption (in case fusee-secondary.bin got corrupted), but I'm gonna be honest and say this looks a lot like a hardware issue to me...particularly since stock is failing to boot, and stock doesn't have anything from atmosphere in play.

Going to close this (since it looks like bit rot of some unknown source), but I would recommend trying true-stock (remove autorcm, or whatever), and if that fails use [an unbricking guide](https://suchmememanyskill.github.io/guides/unbrick/) to restore NAND to known good state, and if true stock *still* fails at that point you're shit out of luck.

It's my expectation true stock will fail to boot (I think RAM issue, tbh), but it's worth trying nonetheless.

closed 2021-07-18 by SciresM

peter765, 2021-07-20:
Thanks for the pointers! I couldn't find anything on this particular issue, so I appreciate the guidance. 

Will do as you say, best of luck!