Atmosphere-NX/Atmosphere #1571: Make Insert ASM Codetype for dmnt Cheats (Perhaps Make This Codetype 0xB)

ReclaimerShawn, 2021-07-19:
## Feature Request: 

The Cheat System (Gecko) for the Nintendo Wii had something known as tbe C2 Codetype, which inserts ASM code into memory. Basically, the code type replaces the original function with a branch, branches to an empty part of memory (that the code handler itself finds), inserts your ASM code there, and then branches back to the line after the original function. 

### What feature are you suggesting?
#### Overview:
The Code Type was as follows:
C2XXXXXX NNNNNNNN
ZZZZZZZZ ZZZZZZZZ
ZZZZZZZZ ZZZZZZZZ
60000000 00000000 

X is the address of the function to replace with the code-handler calculated branch, N was the lines of code to insert (where each line is two functions), Z is the ASM code in bytecode, 60000000 was a NOP in PowerPC (processor architecture of the Wii, purpose was to make the number of functions odd), and 00000000 was to tell the code-handler to branch an instruction after the replaced instruction. 

In ARM, that's basically this:
Address:               Instruction:
XXXXXXXXXX -> B EmptyMemoryAddress 

_EmptyMemoryAddress: 
CODE HERE (All instructions you wrote)
NOP
RET 

I know Code Type 0 can be used to insert ASM, but it doesn't allow you to find empty spaces in memory, branch to them, and branch back automatically. 

#### Smaller Details: 
I even have a suggestion for the format of the codetype. Why not use Codetype 0xB for this, like so:
B00000XX XXXXXXXX
B1YYYYYY
ZZZZZZZZ
D65F03C0 

Where X is the address of the instruction to replace, Y is the number of instructions to insert, Z is the instruction in bytecode (each individual line has one instruction in bytecode, where each line is counted towards YYYYYY), and D65F03C0 is a RET instruction. B0 sets the address and B1 sets the number of instructions to insert. 

#### Nature of Request:  
- Addition: You'd have to make the code-handler find an empty space in memory greater than or equal to the number of instructions you plan to insert, likely inside of the game application itself. You'd also need to make the code-handler branch back to the function 4 bytes away after the first, hopefully without the cheater needing to spell out thaf instruction themselves. 

### Why would this feature be useful?
This feature would be useful because otherwise, a person would have to wade in memory for space in the MAIN.nso to attempt to find any free space. I'm new to ARM and ASM Coding, so maybe this is why, but I myself have not been able to find an empty place in MAIN.nso to insert functions in. Not only that, but space in MAIN.nso is limited, so you could use this function to either write to unused areas of HEAP/BASE or even make it so the instructions are loaded from dmnt cheats itself instead of game memory to prevent writing to game memory.

SciresM, 2021-07-19:
Finding empty space generically isn't actually a solvable problem (since there's only the page alignment space at the end of .text), and dmnt does not have the capability to make data pages executable in the target process.

in other words: "technical issues".

A sane design wouldn't use multiple vm instructions, it would just be one -- I would be willing to consider integrating/improving a code type which did this if presented with a pull request that solved the above technical issues, but yeah gonna close this in the meantime.

SciresM, 2021-07-19:
"make it so the instructions are loaded from dmnt cheat module itself instead of game memory" is not actually possible in the switch's OS btw.

closed 2021-07-19 by SciresM