Atmosphere-NX/Atmosphere #1572: some homebrew apps/games are crashing

n0201, 2021-07-20:
## Bug Report

### What's the issue you encountered?

When I try to start a homebrew game/app my switch crashes in applet mode and the application crashes in normal mode

### How can the issue be reproduced?

Start the emuMMC with Hekate or fusee and then start a application as homebrew menu or start it over album. Then you are going to start Eduke32, Checkpoint or Retroarch (Retroarch doesn't crash all the time). When you use applet mode the switch will crash with error 2144-0001 (0x290). When you did it with a application overridet with homebrew menu the application will crash.

### Crash Report

[](https://gist.github.com/n0201/584c5a74f89cb22db913d51de0fbefe3)

### System Firmware Version

12.1.0

### Environment?

- Hekate v5.5.8 and fusee (im switching between these)
- official release Atmsophere v0.19.5 master fusee was with the 0.19.5 update
- im loading sys-clk and Noexes
- Homebrew software installed: Eduke32, sys-clk manager, Checkpoint, Daybreak, EdiZon SE, Goldleaf, HB App Store, Kosmos Toolbox, NXThemes Installer
- Im using EmuMMC partition based

### Additional context?

-not every Homebrew app crashes, but every homebrew game does

n0201, 2021-07-20:
sorry i forget to left the crash report link. here it is: https://gist.github.com/n0201/584c5a74f89cb22db913d51de0fbefe3

Masamune3210, 2021-07-20:
Piracy is not supported

n0201, 2021-07-20:
> Piracy is not supported

Yeah, I never pirated! I only used apps from the homebrew app store. No pirated stuff I swear!

Masamune3210, 2021-07-20:
Sigpatches and piracy tools count as piracy just as well

n0201, 2021-07-20:
> Piracy is not supported

And I'm a new in this stuff. How do you see that I'm pirating? Is it noexes or Eduke32 (but Eduke32 is an emulator and noexes is turned off all the time. I doesn't even now W what noexes does)

Masamune3210, 2021-07-20:
Anything that requires sigpatches is considered piracy, that includes installing nsp/nsz/xci files

n0201, 2021-07-20:
I'm new pls help I just wanna play eduke and all this stuff. I have no pirated switch games NOTHING! I just wanna play my homebrew games :(. And I just tested checkpoint because I wanted to check if it's also crashing

Masamune3210, 2021-07-20:
Then clean out your SD card and install a stock install of atmosphere from here. No third party packs, no sigpatches, no piracy homebrew like goldleaf

Masamune3210, 2021-07-20:
If the issue persists, make a new issue

closed 2021-07-20 by SciresM