Atmosphere-NX/Atmosphere #1573: Support of 1bit SD issue

karama89, 2021-07-20:
Greetings
I came to know after updating to 12.1.0 that the atmosphere 0.19.5 will not support 1bit SD

Once I lunch my sysMMC from Hekate it goes to black screen.

so is this thing will be fixed in future updates ?
Because I’m in a country where no Switch Repair Shop Available :(

Masamune3210, 2021-07-20:
it is not a software issue, rather a hardware issue. either your SD card reader is loose, the reader is bad, the port on the board is bad, or something else relating to the SD hardware. try opening up the back of the switch and unplugging and replugging the SD card reader in

masagrator, 2021-07-20:
If they are disconnecting reader, then they should already check pins before connecting it back. Pins are very fragile, so all things should be checked with utmost attention so you won't try to disconnect it again unnecessarily.

closed 2021-07-21 by SciresM