Atmosphere-NX/Atmosphere #1576: Homebrew Apps/Games are crashing. Also after a reinstall from atmosphere

n0201, 2021-07-23:
### Bug Report
## What's the issue you encountered?

When I try to start a homebrew game/app my switch crashes in applet mode and the application crashes in normal mode
## How can the issue be reproduced?

Start the emuMMC with Hekate or fusee and then start a application as homebrew menu or start it over album. Then you are going to start Eduke32, Checkpoint or Retroarch (Retroarch doesn't crash all the time). When you use applet mode the switch will crash with error 2144-0001 (0x290). When you did it with a application overridet with homebrew menu the application will crash.
Crash Report

## System Firmware Version

12.1.0
## Environment?

  -   fusee
  -  official release Atmsophere v0.19.5 master fusee was with the 0.19.5 update
  -  im loading sys-clk and Noexes
  -   Homebrew software installed: Eduke32, sys-clk manager, Daybreak, HB App Store,Payload launcher, reboot to payload
  -  Im using EmuMMC partition based

## Additional context?

-not every Homebrew app crashes, but every homebrew game does

n0201, 2021-07-23:
here is the crash report.
https://gist.github.com/n0201/8a8b4d827a9fe0106f0437093f4c7bf4

n0201, 2021-07-23:
ah and i dont use tesla and noexes anymore

Aura314, 2021-07-23:
For Retroarch, hold down R and start a game the same way for checkpoint do not start from the album and piracy is not supported

n0201, 2021-07-23:
ok first, i dont really use checkpoint (i deinstalled it) and i testet it with the r method on a switch game, and this also doesnt work. sometimes retroarch starts a game, but sometimes it crashes. Eduke32 crashes all the time

SciresM, 2021-07-23:
Not an atmosphere issue.

 Homebrew older than (fall 2019) needs to be recompiled to work on 12.0.0/ams 0.19.x.

Update your homebrew to latest versions compiled since then.

closed 2021-07-23 by SciresM