Atmosphere-NX/Atmosphere #1577: black screen

Gaudeu, 2021-07-24:
## Bug Report

### What's the issue you encountered?

After injecting the Atmosphere payload, nothing happens and the screen remains dark.

### How can the issue be reproduced?

Put the nintendo on RCM mode, incect the payload.

### Crash Report

N/A
### System Firmware Version

X.X.X</br>
11.0.0

### Environment?

FUSEE
### Additional context?

N/A

SciresM, 2021-07-24:
"Injecting the atmosphere payload".

Are you sure your switch is unpatched? What happens when injecting hekate?

Gaudeu, 2021-07-28:
yes i'm sure.
I found out the problem was on the micro SD, i just buy another
thx boe

BROSIF85, 2021-07-31:
I am getting a black screen as well, but my issue is different. It happens literally every other boot after the atmosphere and switch logos. It’s like the switch stops booting into emunand. I have tried different memory cards, and reformatting the cards with FAT32 and exFAT. Reloaded all files and built new emunand. All files are up to date. The problem persists. It’s only with emunand every other boot. Never had this problem until 0.19.5. I have an unpatched switch and a chipped V2. It happens on both. Still works. Just annoying that it has to be powered down manually and power on again. Not sure if anyone has had this happen before.

ImNotStealth, 2021-08-01:
For me the Atmosphere logo appears but then the Switch logo doesn't appear. I hadn't used my Switch in a while but I updated it to 12.1.0 so everything should be fine. (I don't think it's the SD card because it worked before)

BROSIF85, 2021-08-02:
After trying a 3rd memory card my issue seems to be resolved. I guess there was an issue with the other 2. I had been using them for a couple of years now, so I guess they could have both went bad at the same time.

ImNotStealth, 2021-08-02:
Actually my issue was fixed by deleting the atmosphere and sept folder :D

closed 2021-08-03 by SciresM