Atmosphere-NX/Atmosphere #1578: Switch 12.1.0 crashing after nintendo switch logo

SquirrelOn0IQ, 2021-07-25:
Not sure what happened but I can't for the life of me get atmosphere to boot, It passes the switch logo and for a split second you can see any alerts like "low battery" and then I get this
[image]

SciresM, 2021-07-25:
1. You didn't use the issue templates

2. Delete your custom theme.

closed 2021-07-25 by SciresM

SquirrelOn0IQ, 2021-07-25:
> 1. You didn't use the issue templates
> 2. Delete your custom theme.

1. How do I delete the theme if I can't boot
2. Sorry, I'm new to github and didn't know about any of the template stuff

Masamune3210, 2021-07-25:
Maybe go read a bit about how things work? Google is your friend....