Atmosphere-NX/Atmosphere #1579: fatal error title id: 010041544d530000 std::abort() called (0xfss)

SquareX12, 2021-07-25:
Hi i'm new to modding, 2 years ago i modded my switch and installed atmosphere. After some months i tried to manually update it with choidujour (i don't remember now if was NX or not) and, when i rebooted it, it shows me this message

A fatal error occurred whem running Atmosphere
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffs)
Report saved to/atmosphere/fatal_errors/report_00000000066dd824.bin

after trying to resolve this problem with other posts of other users, without any success, i just updated the atmosphere version to the last and nothin, still doesn't work, at the end of this post you can find my error file, thanks in advance for your patience and sorry for my bad english.
[report_00000000066dd824.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6874180/report_00000000066dd824.zip)

masagrator, 2021-07-25:
ChoiDojourNX is not supported and even advised to not use it under any circumstances because it's bugged and can install updated with errors.

If you have a backup, restore it and install update using Daybreak which is included with Atmosphere.

SquareX12, 2021-07-25:
I don’t have any backup anymore cuz my old pc destroyed my hdd, and i lost everything and can’t access homebrew so i couldn’t use daybreak and can’t update Inviato da Posta per Windows 10 Da: MasaGratoRInviato: domenica 25 luglio 2021 18:21A: Atmosphere-NX/AtmosphereCc: SquareX12; AuthorOggetto: Re: [Atmosphere-NX/Atmosphere] fatal error title id: 010041544d530000 std::abort() called (0xfss) (#1579) ChoiDojourNX is not supported and even advised to not use it under any circumstances because it's bugged and can install updated with errors.If you have a backup, restore it and install update using Daybreak which is included with Atmosphere.

masagrator, 2021-07-25:
Yeah, I edited my previous post because I was too rushing to conclusions.

So my other question is:
Do you have anything in contents folder?

SquareX12, 2021-07-25:
lemme check... uhm yes i have 12 folder

masagrator, 2021-07-25:
Delete whole contents folder and try again - of course if you are on newest atmosphere.

SquareX12, 2021-07-25:
yes i updated it thinking it was the problem... yes i'm gonna delete it, with tegra wich payload should use?

SquareX12, 2021-07-25:
ok i rebooted the switch and give the same previous error

closed 2021-08-03 by SciresM