Atmosphere-NX/Atmosphere #1580: Panic ID: 010041544D530000

josete2k, 2021-07-28:
If I enable the line in systen_settings.ini

dmnt_always_save_cheat_toggles = u8!0x0

Atmosphere give an error message after the AMS logo

panic occured while running atmosphere, title ID: 010041544D530000

masagrator, 2021-07-28:
You didn't use issue template.

Beside you could upload this system_settings.ini file, report file mentioned in panic error screen and pic of display showing this error.

josete2k, 2021-07-28:
Sorry...

```
[eupld]
; Disable uploading error reports to Nintendo
; upload_enabled = u8!0x0
[usb]
; Enable USB 3.0 superspeed for homebrew
; 0 = USB 3.0 support is system default (usually disabled), 1 = USB 3.0 support is enabled.
; usb30_force_enabled = u8!0x0
[ro]
; Control whether RO should ease its validation of NROs.
; (note: this is normally not necessary, and ips patches can be used.)
; ease_nro_restriction = u8!0x1
; Atmosphere custom settings
[atmosphere]
; Reboot from fatal automatically after some number of milliseconds.
; If field is not present or 0, fatal will wait indefinitely for user input.
; fatal_auto_reboot_interval = u64!0x0
; Make the power menu's "reboot" button reboot to payload.
; Set to "normal" for normal reboot, "rcm" for rcm reboot.
; power_menu_reboot_function = str!payload
; Controls whether dmnt cheats should be toggled on or off by
; default. 1 = toggled on by default, 0 = toggled off by default.
 dmnt_cheats_enabled_by_default = u8!0x0
; Controls whether dmnt should always save cheat toggle state
; for restoration on new game launch. 1 = always save toggles,
; 0 = only save toggles if toggle file exists.
  dmnt_always_save_cheat_toggles = u8!0x0
; Enable writing to BIS partitions for HBL.
; This is probably undesirable for normal usage.
; enable_hbl_bis_write = u8!0x0
; Enable reading the CAL0 partition for HBL.
; This is probably undesirable for normal usage.
; enable_hbl_cal_read = u8!0x0
; Controls whether fs.mitm should redirect save files
; to directories on the sd card.
; 0 = Do not redirect, 1 = Redirect.
; NOTE: EXPERIMENTAL
; If you do not know what you are doing, do not touch this yet.
; fsmitm_redirect_saves_to_sd = u8!0x0
; Controls whether am sees system settings "DebugModeFlag" as
; enabled or disabled.
; 0 = Disabled (not debug mode), 1 = Enabled (debug mode)
; enable_am_debug_mode = u8!0x0
; Controls whether dns.mitm is enabled
; 0 = Disabled, 1 = Enabled
; enable_dns_mitm = u8!0x1
; Controls whether dns.mitm uses the default redirections in addition to
; whatever is specified in the user's hosts file.
; 0 = Disabled (use hosts file contents), 1 = Enabled (use defaults and hosts file contents)
; add_defaults_to_dns_hosts = u8!0x1
; Controls whether dns.mitm logs to the sd card for debugging
; 0 = Disabled, 1 = Enabled
; enable_dns_mitm_debug_log = u8!0x0
; Controls whether htc is enabled
; 0 = Disabled, 1 = Enabled
; enable_htc = u8!0x0
[hbloader]
; Controls the size of the homebrew heap when running as applet.
; If set to zero, all available applet memory is used as heap.
; The default is zero.
; applet_heap_size = u64!0x0
; Controls the amount of memory to reserve when running as applet
; for usage by other applets. This setting has no effect if
; applet_heap_size is non-zero. The default is 0x8600000.
; applet_heap_reservation_size = u64!0x8600000
```
[image]

josete2k, 2021-07-28:
https://www.mediafire.com/file/tdhvk98kv588gcp/report_08d2a7bf.bin/file

The report.bin

masagrator, 2021-07-28:
@josete2k i meant system_settings.ini file, not insides. To check if there are issues with break lines, etc.

josete2k, 2021-07-28:
I only removed the ; before the lina

Using Notepad++

https://www.mediafire.com/file/lmdk4itsspxi4lo/system_settings.ini/file

masagrator, 2021-07-28:
And have you tried to remove space too? Dunno if this can be an issue.

Because setting names should be at the beginning of line. And they are not.

josete2k, 2021-07-28:
Well, the another line is working as weel with the space...

 dmnt_cheats_enabled_by_default = u8!0x0

So I supposed that there isn't mandatory

josete2k, 2021-07-28:
OK... removing the space did the trick.

closed 2021-07-28 by josete2k