Atmosphere-NX/Atmosphere #1581: Improve dns-mitm with the ability to use WiFi without the Internet access.

MAN-biker, 2021-07-28:
## Feature Request

Hi all!

Сan anybody know how one may add the functionality to AMS to use WiFi on a Switch without the necessary of Internet connection, just with a LAN access?

Maybe there's an option to use dns-mitm sysmodule with the right **200 OK** HTTP answer and **X-Organization Nintendo** header.
Here's the discussion with the build of the **bind9**, **yadns** and **Docker**:
[https://gbatemp.net/threads/yadns-nx-yet-another-dns-for-connect-your-switch-to-your-wifi-without-internet-in-a-docker.510239/](https://gbatemp.net/threads/yadns-nx-yet-another-dns-for-connect-your-switch-to-your-wifi-without-internet-in-a-docker.510239/)

I guess there may be much easier solution on a Switch itself, without all that messing around with additional standalone PC/device.

Thanks in advance!

SciresM, 2021-07-28:
This isn't really a dns.mitm task, hosting an http server is definitely out of scope for it.

Probably you could write a custom sysmodule?

closed 2021-07-28 by SciresM

masagrator, 2021-07-28:
https://github.com/misson20000/exefs_patches#nifm-ctest

This is what you are looking for.