Atmosphere-NX/Atmosphere #1583: Error 2162 0002 When Starting up certain games

trextaco9, 2021-08-04:
Been Getting an error 2162 0002 every time I start up certain games like super mario odyssey , I've seen some stuff about deleting 0100000000001000 in content folder in atmosphere and I tried it and it didn't work, Anyone got a solution?

Aura314, 2021-08-04:
Updating Atmospehre to 19.5 and updating your Firmware to 12.1.0.

fincs, 2021-08-04:
Reopen using the issue templates.

closed 2021-08-04 by fincs

trextaco9, 2021-08-05:
> Updating Atmospehre to 19.5 and updating your Firmware to 12.1.0.

well i updated both firmwares and now it's stuck at the logo

Masamune3210, 2021-08-05:
How did you update

trextaco9, 2021-08-05:
> How did you update

used daybreak, preserve settings and fat32 and exfat

Masamune3210, 2021-08-05:
clean sd card of anything that isnt the Nintendo folder to a safe place, and extract a clean download of atmosphere onto the card and try pushing fusee again

trextaco9, 2021-08-05:
I'm able to use the switch now but I'm still getting the error of the first problem on super mario odyssey, and now some of my games are unable to start software. the problems gotten worse.

trextaco9, 2021-08-05:
scratch that all of them

trextaco9, 2021-08-05:
except the ones that were installed before cfw

trextaco9, 2021-08-05:
> clean sd card of anything that isnt the Nintendo folder to a safe place, and extract a clean download of atmosphere onto the card and try pushing fusee again

I now have the first problem of 2162 0002 again everythings updated and working all the games will start up but still do the first error

Masamune3210, 2021-08-05:
reopen issue using the template and state all known information and fill out the template.