Atmosphere-NX/Atmosphere #1584: stuck at logo

trextaco9, 2021-08-05:
## Bug Report

### What's the issue you encountered?

I an error code of 2162 0002 on some of my games

I haven't changed much of anything since i haven't used it as often

### How can the issue be reproduced?

I just open a game and it just errors

### Crash Report

process name : Audio

### System Firmware Version

12.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee primary
- Official release or unofficial build: 19.5 
 
- Do you have additional kips or sysmodules you're loading: no 
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: EmuMMC

THZoria, 2021-08-05:
I also had this problem, and after checking in more detail, it was related to the TriPlayer sysmodule for the background music player
(problem affecting Minecraft Dungeons)

trextaco9, 2021-08-05:
> I also had this problem, and after checking in more detail, it was related to the TriPlayer sysmodule for the background music player
> (problem affecting Minecraft Dungeons)

Oh my god you were right

trextaco9, 2021-08-05:
> I also had this problem, and after checking in more detail, it was related to the TriPlayer sysmodule for the background music player
> (problem affecting Minecraft Dungeons)

Oh my god it even fixed some of the games

closed 2021-08-05 by trextaco9