Atmosphere-NX/Atmosphere #1589: Panic Occurred while running Atmosphere (0xFFE)

bluetypewriter, 2021-08-10:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
Upgrade to Atmosphere 0.19.5</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
None, other than the upgrade.  Upgrade was completed as a clean install</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>
N/A</br>

### How can the issue be reproduced?
Boot into Atmosphere.  Additionally, I placed a copy of hekate_ctcaer_5.5.8.bin in /atmosphere and renamed it to reboot_payload.bin.  After error occurs, hekate will load.  I tried resetting Arch bit but issue remains

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

</br>
[report_0c468170.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6958818/report_0c468170.zip)

### System Firmware Version

12.1.0</br>
</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Attempted with both.  Issue is the same regardless
- Official release or unofficial build: 
  - Official 0.19.5
- Do you have additional kips or sysmodules you're loading:
- No
- Homebrew software installed: 
- Daybreak
- EmuMMC or SysNAND:
  - EmuMMC, partition based

### Additional context?

- Additional info about your environment:
- SysNAND works without issue

bluetypewriter, 2021-08-12:
Issue is not with atmosphere.  Corrupted EMUNAND.  Discussed in discord and decided to export saves and rebuild a new EMUNAND

closed 2021-08-12 by bluetypewriter

einalrem, 2021-08-15:
I have the same problem now. Im using SysNand. But still get this message. 
Panic occurred while running Atmosphere,
Title ID: 0100000000000002b
Error: std: abort (0xFFE)