Atmosphere-NX/Atmosphere #1590: Can't launch very specific titles after updating to 0.19.5

LeyendaV, 2021-08-12:
## Bug Report

### What's the issue you encountered?

Everything was working perfectly when I was using 0.19.4
Since I updated to 0.19.5, custom forwarders are not working, as well as YouTube. When it comes to actual games, the only one pesenting problems is Skyrim (yes, a legic copy).
I'm assuming is a patches problem, since forwarders need them, YouTube uses one to be able to use the app from the emunand, and Skyrim uses one too to launch multiple mods.

### How can the issue be reproduced?

By trying to lauch a forwarder or any app that requires a custom sigpatch on 0.19.5

### Crash Report

There's no crach report. You only get the "The software was closed because an error occurred" message when launching the title and then it closes.

### System Firmware Version

12.0.3

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: M.19.5
- Do you have additional kips or sysmodules you're loading: The ones mentioned above.
- Homebrew software installed: Other sysmodules running are Emuiibo and Tesla
- EmuMMC or SysNAND: Emunand (File)

### Additional context?

- Additional info about your environment: Everything was working perfectly on 0.19.4, then stop working on 0.19.5. In both cases I was (and am) using HOS 12.0.3

SciresM, 2021-08-12:
Piracy (including all custom content installation/"forwarders") is not supported.

closed 2021-08-12 by SciresM

LeyendaV, 2021-08-12:
The only custom forwarder I use is Checkpoint tho