Atmosphere-NX/Atmosphere #1591: A fatal error occurred when running Atmosphere. Program ID : 0100000000000003

Tumnus108, 2021-08-12:
## Bug Report

Hello guys,
I encountered this bug and couldn't find any solution on the existed bug reports.
Could you help me with this one please ?
Thanks by avance!

### What's the issue you encountered?

While launching a game we accidentally clicked on update the game. Then the Atmosphere crashed and the switch booted back to Hekate.
I tried to launch again the CFW but an error message occurred (not the same as this one but I can't remember which error it was)
I then tried to update atmosphere, the sigpatches and Hekate.
I tried to boot again to Atmosphere and this error occurred.

### How can the issue be reproduced?

Launch Hekate.
Launch Atmosphere on the CFW.
Error occurs 

### Crash Report

[report_0000000007d5d030.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6977835/report_0000000007d5d030.zip)

blob:file:///313c93ee-64c5-413d-9124-7b86baa90f34

### System Firmware Version

Firmware 11.0.1

### Environment?

- Atmosphere launched by hekate
- Official release version AMS 0.19.4.
- No additional kips or sysmodules
- Homebrew software installed: Goldleaf, Kosmostoolbox, Kosmosupdater, NS-usblaoder
- EmuMMC partition-based

### Additional context?

- Additional info about your environment: N/A

Masamune3210, 2021-08-12:
Piracy is unsupported

Tumnus108, 2021-08-13:
> Piracy is unsupported

Hello @Masamune3210, I'm not sure to understand what is the source of my error and what I should do to correct it... Thanks

SciresM, 2021-08-13:
You have the archive bit set on atmosphere/contents subdirectories, and also haven't upgraded properly to atmosphere 0.19.x.

Delete atmosphere directory, extract latest release zip, make sure archive bits aren't set.

closed 2021-08-13 by SciresM

Tumnus108, 2021-08-13:
> You have the archive bit set on atmosphere/contents subdirectories, and also haven't upgraded properly to atmosphere 0.19.x.
> 
> Delete atmosphere directory, extract latest release zip, make sure archive bits aren't set.

@SciresM, thanks a lot, it worked perfectly ! 

I don't understand why it did not work before since the first thing I did was to reinstall Atmosphere, but the important thing is that it is fixed... Thanks :)