Atmosphere-NX/Atmosphere #1592: Fatal error ocurred when running Atmosphére Title ID: 010000000000002B

einalrem, 2021-08-15:
## Bug Report

### What's the issue you encountered?

I'm on 12.1.0, with atmos at 0.19.5, I got this message when I inject payload in tegra using the newest version of hekate </br>
Panic occured while running Atmosphere. 
Title ID: 010000000000002B
Error : std: :abort <0xFFE>
Report saved to /atmosphere/fatal_errors/report_08d0bb22.bin"

### How can the issue be reproduced?

After the atmosphere logo appeared</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/fatal_reports"
[63ed27475ac7c9aa56e8c3a74122135b-2d054ddeb211ce3fcb58ff89a4cc552d334f1e2e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6988240/63ed27475ac7c9aa56e8c3a74122135b-2d054ddeb211ce3fcb58ff89a4cc552d334f1e2e.zip)

### System Firmware Version

12.1.0</br>

### Environment?

- I'm using the latest version of hekate 5.5.8 and official release version atmosphere 0.19.5 
- SysNAND

### Additional context?

- Additional info about your environment:
- [ N/A]

SciresM, 2021-08-17:
Neither of the files you posted are the crash report you mention in the issue.

Please upload report_08d0bb22.bin, else I'll close this in 24 hours.

Situro, 2021-08-17:
I got the same problem, someone in the discord helped me with this error
In Hekate go in Tools > Arc Bit (bottom right) > Fix Archive Bit
See if that sets/unsets anything, and try to launch Atmosphere again

Akaiahh, 2021-08-17:
Hi @SciresM 
I got the same error this morning: same hekate & atmosphere version, but this happens when starting emuMMC from Hetake (on version 12.0.0)
Do you want me to create a separated issue to upload my "report.zip" or I can send it here ?

BTW, I was running with hekate 5.5.5 and Atmosphere 0.19.0 (emunand 12.0.0) for months without any trouble. This happend on a reboot after an atmosphere crash.

Even after updating  hekate&atmosphere to the latest version, this still occurs.
Regards.

PS
I tried the "fix archive bit" and the "remove /saves/80000000000000d1 in system partition". No good results from this.

SciresM, 2021-08-17:
Upload the generated report file (from /atmosphere/fatal_errors) and link it in a comment here

Akaiahh, 2021-08-17:
[report.zip](https://github.com/Akaiahh/logs/raw/main/report_0b72849c.zip)

SciresM, 2021-08-17:
@Akaiahh You have a corrupt PRODINFO/system save.

Follow the instructions/directions in #1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

Akaiahh, 2021-08-18:
Hum...
I do not have a recent "PRODINFO" backup (shame on me). Can I use an old one (9.0.0 ? I can't try right now, this old backup is "far" from me).
NxNandManager only allow me to wipe personnal data of PRODINFO partition, but it did not work. (combined with removing 8000000000000050 and 8000000000000052 from SYSTEM) is that enougth ?

Can you confirm me, with the [following error log](https://github.com/Akaiahh/logs/raw/main/report2_0b0ecba8.zip), that is still the same root cause (PRODINFO/system save corrupted) ?

If yes, I will try to export my saves from the current emuMMC then recreate a fresh emummc and proper backups.

Thx
Akaiah

SciresM, 2021-08-18:
PRODINFO does not change on system update (or ever), so yes the old backup will work.

Wiping personal data won't help -- having personal data wiped is in fact part of the problem.

Yes that error log is same issue as I said before.

Akaiahh, 2021-08-18:
It works.

I was able to retore the PRODINFO partition from an old backup @version 9.0.1
The [#1256](https://github.com/Atmosphere-NX/Atmosphere/issues/1256) says we have to also delete 8000000000000050 and 8000000000000052 from SYSTEM partition. By doing this, I was stuck on a black screen after the atmosphere logo, no fatal error anymore. And none of these files where recreated on startup.

To make it work, I also had to recover 8000000000000050 and 8000000000000052 from the same old backup.

to sum up, I needed to restore:
- PRODINFO partition
- 8000000000000050 and 8000000000000052 to SYSTEM partition.

Thank you for your help.
Regards.

closed 2021-08-22 by SciresM

Neph81, 2022-03-20:
> It works.
>
> I was able to retore the PRODINFO partition from an old backup @Version 9.0.1 The [#1256](https://github.com/Atmosphere-NX/Atmosphere/issues/1256) says we have to also delete [...]

Thank you.  This fixed the issue I was having with my system.