Atmosphere-NX/Atmosphere #1595: Panic Error for atmosphere everytime I load up Smash

LeoOfTheCosmos, 2021-08-22:
## Bug Report

### What's the issue you encountered?

I was just loading up as usual one day, realized I was not LANNED in; shut down smash to reload and LAN in and the problem has persisted since.</br>
I have not made any changes to atmosphere as I know of. I have fixed the bit archive in attempt to repair but has not worked for me. </br>

### How can the issue be reproduced?
Everytime I load up Smash 

I place RCM in side, enter RCM mode(with SD card in), already have TegraRCMGUI loaded, connect with cable, inject my payload. I load up Hekate bootloader. Launch it. select sysMMC. Remove RCM and cable, go to load up Smash bros. Will begin loading, then panic error crash. It has successfully loaded about once out of 20 tries as of now. Usually I just have to reboot and the error ceases, it does not as of yesterday. 

### Crash Report
Title ID:010041544D530000
Error: std: :abort (0xFFE)

### System Firmware Version

12.1.0Switch  5.5.8Hekate</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: I believe official
  - I acquired it through this link from a guide of a trusted person https://www.sdsetup.com/console?switch#atmosphere;atmos_musthave;hekate_musthave;hbmenu;hekate;

- Do you have additional kips or sysmodules you're loading: I apologize I'm new and would need some guided direction as to what this means. N/A
- Homebrew software installed: N/A
- EmuMMC or SysNAND: I always launch CFW (sysMMC)
  

### Additional context?

- Additional info about your environment:
I mod and reskin characters in Smash but it hasn't really been an issue until now. I am glad to provide any possible assistance at all in helping to make this easier. I apologize if I filled this out incorrectly; I did attempt with my best effort.
Also, My voltage goes into the negative -400 a little after booting up.. is that a possible issue? Just trying to save time if it is.

[image] Contents of my atmosphere folder

[report_097b8745e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7026386/report_097b8745e.zip)  
All my fatal errors

[image]
What mods downloaded and the location is my own folder I was designated to make

Please let me know if there is anything else at all needed.

SciresM, 2021-08-22:
Basically all the reports in your error folder are FS result 2060 (sd card device status error).

It looks like your SD card is fucked. 

Try a different SD card?

closed 2021-08-22 by SciresM

LeoOfTheCosmos, 2021-08-22:
Alright thank you so much. Should I just copy and paste all content from this card to the next and be all set? Also would you prefer I comment here or elsewhere