Atmosphere-NX/Atmosphere #1598: after updating  my mariko version with the latest atomosher version v20   when i lunch games it become unstable  it heat up qucik a huge fps drop

skywalker454, 2021-08-29:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

skywalker454, 2021-08-29:
kindly  your support is highly appreciated   the device become boot  up quick but the  problem when open  games it freez and heat up quuck   if  that r  related  ram over clocking i  have  notices that the ram speed become  1666 mhz

masagrator, 2021-08-29:
You didn't use properly issue template.
Beside use full sentences with dots. Your comment is hard to read.

skywalker454, 2021-08-29:
okay sorry iam still new in github  i was trying to say that my  mariko version after  it updated with the   latest  atmosphere v20 it start to heat up quick  when i open games some times freeze when i roll back to the  pervious version 19.5 it works fine what the cause   of  the issue ? 
thanks in advance

lowburger, 2021-08-29:
Fill issue in properly... it's dumpster fire

orbstar8, 2021-08-29:
In summary I will tell you what happens, because the same thing happens to me, I have a mariko switch, when I put the console in standby mode, for some reason, the use of the GPU does not drop below 40%, and the system both the games go with too many performance problems and even graphics !!  FW 12.0.3 EmuNAND

skywalker454, 2021-08-29:
> In summary I will tell you what happens, because the same thing happens to me, I have a mariko switch, when I put the console in standby mode, for some reason, the use of the GPU does not drop below [...]

  yeaa    the same thing happens to me  if i  left it in standby   mode i found it fired up and lags alots  is that due  to  ram over clocking speed   it  increase up to  1666mhz ?

skywalker454, 2021-08-29:
> In summary I will tell you what happens, because the same thing happens to me, I have a mariko switch, when I put the console in standby mode, for some reason, the use of the GPU does not drop below [...]

> yeaa the same thing happens to me if i left it in standby mode i found it fired up and lags alots is that due to ram over clocking speed it increase up to 1666mhz ?
  all i concern  about will   that affects me  in an y next official  update by atmosphere ? :(

digglife, 2021-08-29:
If you really want the maintainers to help on this issue, at least make some effort to format the bug report as the template guides you. The basic manner for raising issue: proper format and information as detailed as possible.

skywalker454, 2021-08-29:
> If you really want the maintainers to help on this issue, at least make some effort to format the bug report as the template guides you. The basic manner for raising issue: proper format and information as detailed as possible.

i uploads attached files  is that okay with the post

SciresM, 2021-08-29:
This issue is incoherent.

That said, it sounds like your switch (lite?) has some problems with lag due to memory defaulting to 1600MHz (since we train it to that point) or something?

I guess these are my questions.
1. Does the issue occur booting AMS 0.20.0 via hekate fss0 instead of fusee-primary?
2. Does the issue occur booting without any custom system modules? Please post a screenshot of your /atmosphere/contents folder, and make sure there is nothing in it when testing.

I guess potentially we'll need to only train switch lites to 1331MHz, depending on whether this is common for lites.

closed 2021-08-29 by skywalker454

SciresM, 2021-08-29:
Sigh, okay you closed this and opened a duplicate. I'll go copy over my comment.