Atmosphere-NX/Atmosphere #1599: Switch Lite game's frame rate drop, lag, hang after upgrade to 0.20.0

parnz, 2021-08-29:
### What's the issue you encountered?
After upgraded from 0.19.5 to 0.20.0 the boot time is extremely decreased **WOW**
but the system is lag, delay or not responsive at all.
Ex. 
- Long press the power botton and use the dpad to select menu items, Notice the delay after press the dpad and the hilighted menu item.
- game frame rate clearly noticeable decreased even cut scene video.
- The worst case is Switch not response, I've holding down the power button to force shutdown.

### How can the issue be reproduced?
Upgrade to 0.20.0

### Crash Report
n/a, unfortunately the sdcard was formatted to rollback to 0.19.5
### System Firmware Version
12.1.0

### Environment
- What bootloader?
 _hekate 5.6.0_
- Official release or unofficial build?
 _Atmosphere Official release 0.20.0_

- Do you have additional kips or sysmodules you're loading:
 _n/a (overlay menu?)_
- Homebrew software installed:
 _EdiZon, tinfoil, chiaki, aio-switch-updater,emuiibo_
- EmuMMC or SysNAND:
_EmuMMC (file-based)_

### Additional context?
- It's a fresh installed sdcard and newly created EmuNAND.
- Signature patches were applied.

masagrator, 2021-08-29:
```
tinfoil
signature patches
```

Piracy is not supported.

skywalker454, 2021-08-29:
i got  the  same  problem of  his in my mariko version

parnz, 2021-08-29:
Hoping some good guy suffering this issue and report.
Well, I'm closing this issue then.

closed 2021-08-29 by parnz

skywalker454, 2021-08-29:
Why you closing it

skywalker454, 2021-08-29:
I hope is there's a  solution to it it's not solved yet will that affect on
the other official updates in the future ?