Atmosphere-NX/Atmosphere #1600: mariko version heated up  after    upgrading  from  version  19.5 the latest   version  of atmosphere 20

skywalker454, 2021-08-29:
What's the issue you encountered?
After upgraded from 0.19.5 to 0.20.0 the boot time is extremely decreased 
but the system is lag, delay or not responsive at all.
Ex. like   left  it on the stand by mood and i figured out it heated up alot and  horrible stuttering in the game 

Long press the power botton and use the dpad to select menu items, Notice the delay after press the dpad and the hilighted menu item.  one the lag examples 

game frame rate  is unplayable at all 
The worst case is Switch not response, I've holding down the power button to force shutdown.
How can the issue be reproduced?
Upgrade to 0.20.0

Crash Report
n/a, unfortunately the sdcard was formatted to rollback to 0.19.5

System Firmware Version
12.1.0
[image]

Environment
What bootloader?
  it  boot directly to the swtich   it works fine with the  pervious  19.5   
that`s the boot loader ini files i got in  my boot loader  folder  
[image]

Official release or unofficial build?
Atmosphere Official release 0.20.0

Do you have additional kips or sysmodules you're loading:
n/a (overlay menu?)
i got edizon   and tesla  and they are working fine in the pervious 19.5 version 

Homebrew software installed:
EdiZon, tinfoil, chiaki, aio-switch-updater,emuiibo  day break   goldeleaf 

EmuMMC or SysNAND:
SysNAND  (file-based)

Additional context?
 your  help is highly appreciated thanks in advance

SciresM, 2021-08-29:
It sounds like your switch (lite?) has some problems with lag due to memory defaulting to 1600MHz (since we train it to that point) or something?

I guess these are my questions.

1. Does the issue occur booting AMS 0.20.0 via hekate fss0 instead of fusee-primary?
2. Does the issue occur booting without any custom system modules? Please post a screenshot of your /atmosphere/contents folder, and make sure there is nothing in it when testing.

I guess potentially we'll need to only train switch lites to 1331MHz, depending on whether this is common for lites.

skywalker454, 2021-08-29:
yeaa it works perfectly fine with  that speed  1331 mhz  ram  on version  19.5   
[image]
[image]
[image]
[image]

skywalker454, 2021-08-29:
at the first i thought in first that it  could that happen due to memory speed  over clock as i  found it after  a couble i of minutes afer lunching game   it start to heat up quick even with any game

SciresM, 2021-08-29:
You did not post /atmosphere/contents on your SD, or answer either of my questions.

skywalker454, 2021-08-29:
[image]

skywalker454, 2021-08-29:
i used  the primary fuse that is what in my sd card

skywalker454, 2021-08-29:
i  don`t use  hekate  it`s just boot directly to atmosphere

skywalker454, 2021-08-29:
what  the folder content got  many folders when i opened it  what these files refers too i thought it something releated to atmosphere files  and  it was in  the pervious  version of  19.5

parnz, 2021-08-29:
@skywalker454 you need to upgrade to 0.20.0 again and then provide any info/picture as requested by @SciresM 
Not your current 0.19.5

zemixe, 2021-08-29:
@parnz if you have a Switch Lite too please provide the required information in this issue.

parnz, 2021-08-29:
> @parnz if you have a Switch Lite too please provide the required information in this issue.
If @SciresM allowed me. I will do.

skywalker454, 2021-08-29:
> @skywalker454 you need to upgrade to 0.20.0 again and then provide any info/picture as requested by @SciresM
> Not your current 0.19.5

 okay i will do it

skywalker454, 2021-08-29:
[image]
[image]

skywalker454, 2021-08-29:
i  used the  hekate version  hekate_ctcaer_5.6.0 this time

skywalker454, 2021-08-29:
the pics i provided i took from my sd card   as  a backup from my hdd driver but that currently the files  found  in my sd card now 
after the update

skywalker454, 2021-08-29:
[image]

skywalker454, 2021-08-29:
guys   i posted all the  pics  you asked me  your  help is highly appreciated iam just what concerns me that i  won`t  be able to update switch atmosphere in future   :(

skywalker454, 2021-08-29:
can you make a version  with   the normal  ram speed  could be the   issue  ?

SciresM, 2021-08-29:
fwiw another user has confirmed this on his lite on discord...you aren't a native english speaker I think so I've been having him do some tests for me.

Don't worry, I'll try to fix it, and if I can't I'll at the very least add an option to disable boot-memory training on switch lite.

Also, you can just boot using hekate instead of fusee.

SciresM, 2021-08-29:
Double fwiw, it appears this is happening (exclusively(?)) on DRAM ID 12 (Samsung Hoag).

A user with a lite on DRAM ID 14 has reported no issues in several weeks of usage.

skywalker454, 2021-08-29:
so  what you means  that  it  occurs on some  specific switch hardware devices like mine ?

Laurentiu-85, 2021-08-29:
I have same problem on switch lite on latest AMS when i boot via fuse-primary .if i launch with hekate fss0 work very vell.

skywalker454, 2021-08-29:
What the best to  fuse to boot the primary or the other one ?

vicoy032193, 2021-08-29:
I have the same issue. switch lite Dram 14. im booting using hekate fss0. so i decided to downgrade for now so i can play any game smoothly.

SciresM, 2021-08-29:
Yes, a solution has been identified for this issue which in truth affects all lites. It will take me a day or two though.

parnz, 2021-08-30:
My Switch Lite have DRAM ID 19 also have this issue.

SciresM, 2021-08-30:
Yes, as mentioned all lites affected.

commit c5edb031 (2021-08-30): fusee: restore DRAM to 204MHz before starting main cpu on Mariko (closes #1600).

Nintendo's Mariko tables result in trained frequency of 1599999 instead of 1600000.

PCV checks for rate == 1600000 exactly, when doing EMC init.

Thus EMC init does not succeed if we are trained to 1600000.

PCV has a fudge factor of 1000 used in SetEmcDvfsFreq, but this is not used in InitEmcDvfs.

This failure means that PCV cannot change rate back to 204MHz before sleep, and then after
wake extremely degraded performance is observed.

Restoring DRAM to 204MHz before boot causes EMC init to succeed/fixes perfo [... cut]
closed 2021-08-30 by SciresM