Atmosphere-NX/Atmosphere #1602: [NXBOOT] Failed to run TSEC firmware -6

myeuropeanPAL, 2021-08-30:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

When booting into emuMMC with atmoshphere 0.20.1 I receive the following error message:
[NXBOOT] Failed to run TSEC firmware -6 d0d0d0d0 d0d0d0d0!

### How can the issue be reproduced?

Booting into emuMMC with hekate v5.6.0 & Nyx v1.0.6

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

myeuropeanPAL, 2021-08-30:
I don't think so, atmosphere-0.19.5 and was booting just fine with hekate v5.6.0 & Nyx v1.0.6

SciresM, 2021-08-30:
"0.19.5" upgrade to latest 0.20.1? Does boot succeed using fusee primary directly instead of hekate, or booting directly via hekate instead of fusee primary?

could you post a screenshot of the console info -> fuses page in hekate if the issue persists?

Ammako, 2021-08-31:
Just to clarify, what they're saying is that the error happens when booting with 0.20.1, but doesn't happen when booting with 0.19.5

SciresM, 2021-08-31:
There've been a few reports of this...all reporters have been using gateway's payload loader, which results in an unexpected boot environment.

If you're using that, switch to spacecraft-nx or whatever.

If you're not, we can re-open the issue/etc.

closed 2021-08-31 by SciresM