Atmosphere-NX/Atmosphere #1606: [NXBOOT] Failed to run TSEC firmware -6 with version 0.20.1

NathanIfinity, 2021-09-01:
[ If any section does not apply, replace its contents with "N/A". ]

[ Lines between [ ] (square brackets) should be removed before posting. ]

[ * ]

[ Note: If the bug or crash you encountered is related to; ]

[ - software used to make "backups", ]

[ - software explicitly distributed for piracy, etc ]

[ then contributors will not provide support for your issue and your issue will be closed. ]

What's the issue you encountered?
When booting into atmoshphere with 0.20.1 I receive the following error message:
[NXBOOT] Failed to run TSEC firmware -6 d0d0d0d0 d0d0d0d0!

How can the issue be reproduced?
Booting into Atmoshpere with hekate v5.6.0 & Nyx v1.0.6

Crash Report
/atmosphere/crash_reports folder does not exist

System Firmware Version
12.1.0

Environment?
What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate, then Fusee Primary, also tried booting Fusse primary directly, still smae message

Official release or unofficial build: official release hekate v5.6.0 & Nyx v1.0.6

Do you have additional kips or sysmodules you're loading: No

EmuMMC or SysNAND: EmuMMC, file-based : SysNAND

Additional context?
Issue only started when I updated to 0.20.1 was working fine with the old version:  0.19.5
put  0.19.5 back and everything is back like it was

I use the loader I got when I first got my Sxos to load Hekate or fusee primary
I saw on other issue you recommend spcaecraft-nx , but not sure exactly what it is.

thanks

SciresM, 2021-09-01:
Are you on a patched console? What do you use to launch hekate/atmosphere?

SciresM, 2021-09-01:
Ah "I use the loader I got when I first got my Sxos"

yeah this is gateway's loader.

Not supported, use spacecraft-nx.

closed 2021-09-01 by SciresM

NathanIfinity, 2021-09-01:
day 1 switch, unpatched, so something changed between 0.19.5 and 0.20.1 ?? because it worked fine before
what exactly is spacecraft-Nx? to I need to replace my loader ?

SciresM, 2021-09-01:
If you're unpatched, just send fusee or hekate directly via the rcm bug.

the issue is that you're using gateway's loader.