Atmosphere-NX/Atmosphere #1610: Sysmmc serial blanking not working on AMS 1.0.0

Quotad, 2021-09-07:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Serial blanking with exosphere.ini not working on Atmosphere 1.0.0

### How can the issue be reproduced?

[ * ]</br>
Copied exosphere.ini from config_templates folder to sd root, set blank_prodinfo_sysmmc=1, checked serial number in system settings and it is not blanked</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>N/a no crash</br>

### System Firmware Version
12.1.0

### Environment?

Chainloaded fusee.bin via hekate 5.6.0. 
Self compilation of custom firmware. 
Sysnand based
Only homebrew installed is Tinfoil

Quotad, 2021-09-07:
Here's the contents of my exosphere.ini
```
# Key: debugmode, default: 1.
#  Desc: Controls whether kernel is debug mode.
#   Disabling this may break Atmosphere's debugger in a future release.

# Key: debugmode_user, default: 0.
#  Desc: Controls whether userland is debug mode.

# Key: disable_user_exception_handlers, default: 0.
#  Desc: Controls whether user exception handlers are executed on error.
#  NOTE: This will cause atmosphere to not fail gracefully.
#   Support may not be provided to users tho disable these.
#   If you do not know what you are doing, leave them on.

# Key: enable_user_pmu_access, default: 0.
#  Desc: Controls whether userland has access to the PMU registers.
#  NOTE: It is unknown what effects this has on official code.

# Key: blank_prodinfo_sysmmc, default: 0.
#  Desc: Controls whether PRODINFO should be blanked in sysmmc.
#   This will cause the system to see dummied out keys and
#   serial number information.
#  NOTE: This is not known to be safe, as data may be
#   cached elsewhere in the system. Usage is not encouraged.

# Key: blank_prodinfo_emummc, default: 0.
#  Desc: Controls whether PRODINFO should be blanked in emummc.
#  NOTE: This is not known to be safe, as data may be
#   cached elsewhere in the system. Usage is not encouraged.

# Key: allow_writing_to_cal_sysmmc, default: 0.
#  Desc: Controls whether PRODINFO can be written by homebrew in sysmmc.
#  NOTE: Usage of this setting is strongly discouraged without
#    a safe backup elsewhere. Turning this on will also cause Atmosphere
#    to ensure a safe backup of calibration data is stored in unused
#    mmc space, encrypted to prevent detection. This backup can be used
#    to prevent unrecoverable edits in emergencies.

# Key: log_port, default: 0.
#  Desc: Controls what uart port exosphere will set up for logging.
#  NOTE: 0 = UART-A, 1 = UART-B, 2 = UART-C, 3 = UART-D

# Key: log_baud_rate, default: 115200
#  Desc: Controls the baud rate exosphere will set up for logging.
#  NOTE: 0 is treated as equivalent to 115200.

# Key: log_inverted, default: 0.
#  Desc: Controls whether the logging uart port is inverted.

[exosphere]
debugmode=1
debugmode_user=0
disable_user_exception_handlers=0
enable_user_pmu_access=0
blank_prodinfo_sysmmc=1
blank_prodinfo_emummc=0
allow_writing_to_cal_sysmmc=0
log_port=0
log_baud_rate=115200
log_inverted=0
```

borntohonk, 2021-09-07:
`ShouldUseBlankCalibrationBinary` does look to be missing, so that'd probably cause that.

SciresM, 2021-09-07:
@borntohonk not sure what you mean by "missing".

fusee definitely tries to set the flag: https://github.com/Atmosphere-NX/Atmosphere/blob/master/fusee/program/source/fusee_setup_horizon.cpp#L626

And usage of that flag hasn't changed in forever...

closed 2021-09-07 by SciresM

SciresM, 2021-09-07:
@Quotad configuration was being overwritten due to emummc coming later in the INI, much appreciated for the report.

Download the fixed pre-release build, it no longer reproduces for me in that.

Quotad, 2021-09-07:
@SciresM Yep updated and all working now, thanks for the quick fix :)

commit 44beeecc (2021-09-07): fusee: fix prodinfo blanking flag detection for sysmmc (closes #1610)