Atmosphere-NX/Atmosphere #1611: Fatal error starting atmosphere?

Retroken101, 2021-09-07:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. 
I started up my switch like normal and it gave me this: "A fatal error occurred when running Atmosphere.
Title ID: 010000000000002b
Error Desc: std::abort() called (0xffe)"]</br>
[ Did you make any changes related to Atmosphère itself? None! None at all!]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ it can't be reproduced whenever I try to use my RCM loader, which I've always used to get into my switch, and start up my switch like normal.]</br>
[ put switch into rcm mode to use my switch, nothing else. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

0.12.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: first it was fusee. Then I tried using hekate.
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ think so?]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. Was using emummc, it's partition based I believe.]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-09-07:
1) you did not fill out the issue templates correctly 
2) upload the fatal error report file that was saved.

Retroken101, 2021-09-07:
Sorry about that. Stand by. I'm new to this and it wouldn't let me post what I was trying to post.

Retroken101, 2021-09-07:
[df82e618d8136b75c26d816c86ff58b5-e9460356b0f91e55a292080486c40042b5d4ec2d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7124720/df82e618d8136b75c26d816c86ff58b5-e9460356b0f91e55a292080486c40042b5d4ec2d.zip)

Retroken101, 2021-09-07:
did i do that right?

Retroken101, 2021-09-07:
the template is a tad bit confusing. would you be so kind as to tell me what all info you need? i will be glad to give you anything you need, i just dont know how to abide by the template. just a bit confusing to me, is all. otherwise, id follow it to a T. this issue has been one of the worst ive faced.

SciresM, 2021-09-07:
No, that error is unrelated to the one I want you to upload.

When the screen you describe ("abort occurred") shows, it will tell you the filename it saved information to. Upload that file, please.

Also note that this issue tracker is for bugs, not for support.

Retroken101, 2021-09-07:
[report_093c080e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7124760/report_093c080e.zip)

Retroken101, 2021-09-07:
like this, sir?

SciresM, 2021-09-07:
You have a corrupt PRODINFO/system save.

Follow the instructions/directions in #1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

closed 2021-09-07 by SciresM

Retroken101, 2021-09-08:
I tried using the guide but got confused on where to find those files to delete. I'm sorry if I'm being a pest. I just want to really fix this and get back to gaming...