Atmosphere-NX/Atmosphere #1612: Fatal Error when launching Atmosphere through Hekate

Lickitahm, 2021-09-08:
## Bug Report

### What's the issue you encountered?

I am receiving a Fatal Error when trying to launch atmosphere after updating to 1.0.0.

### How can the issue be reproduced?

When trying to launch the CFW (SYSMMC) option on Hekate. Using the new Fusee manages to launch CFW.

### Crash Report

[report_06ae03d0.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7125104/report_06ae03d0.zip)

### System Firmware Version

12.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 
  - 1.0.0 Pre-release
- Do you have additional kips or sysmodules you're loading: Tesla and Emuiibo
- Homebrew software installed: Besides the .nro ones, I don't have any I can think off
- EmuMMC or SysNAND:
  - SysNAND

### Additional context?

- Additional info about your environment:
- Launching through Fusee works, this is the error that I receive when trying to launch through Hekate:
(Panic ocurred while running Atmosphere.
Title ID: 010041544D530000
Error Desc: std::abort(0xffe) )

Lickitahm, 2021-09-08:
Sorry for the reopen, any idea on how to update my Hekate config? I'm pretty new with that, didn't know I had to update it to use the new pre-release

Masamune3210, 2021-09-08:
This isnt a support forum, there are places for that. Go look at the latest release of hekate for examples of how to set up the config