Atmosphere-NX/Atmosphere #1615: Daybreak doesn't work on 1.0.0

noirscape, 2021-09-10:
## Bug Report

### What's the issue you encountered?

Updated my brother's switch software to latest version; fully "clean" version of Atmosphere. He has an emuNAND on 11.0.1, I wanted to update it to 12.0.1 with provided daybreak. (Software used to dump is TegraExplorer but should be irrelevant.

### How can the issue be reproduced?

1. Boot up lower emuNAND on Atmosphere 1.0.0 with 12.0.1 dump on SD, read to install.
2. Open up Daybreak.
3. Get "Outdated Atmosphere version. Daybreak requires Atmosphere 0.14.0 or later".
4. Daybreak exits

### Crash Report

This is not a crash.

### System Firmware Version

11.0.1

### Environment?

- Booted with Hekate FSS0 (5.6.0).
- Official release build of Atmosphere (hash is bfeba7c1).
- No additional kips or sysmodules loaded (clean boot)
- partition based emuNAND.

### Additional context?

- Completely clean Atmosphere + Hekate installation (all are on latest).
- Most likely guess is something to do with the below line.

https://github.com/Atmosphere-NX/Atmosphere/blob/0c596e682f5c825e02831c87915093f97b3c12ea/troposphere/daybreak/source/ui.cpp#L1248

SciresM, 2021-09-10:
That line is fine.

I cannot reproduce this. I just successfully used daybreak on ams 1.0.0, with no issue.

Closing, because this is almost certainly a user configuration issue.

closed 2021-09-10 by SciresM

3DSBricker, 2021-09-12:
@SciresM It is not, I am getting the exact same error. Tried both fss0 with the new package3 way and Fusee.bin, I'm sure this is because of the new Atmosphère 1.0.0 and probably because of the new version name.

3DSBricker, 2021-09-17:
I haven't changed anything, but Daybreak suddenly works now. Don't think it's a Daybreak error/problem, but we both had the same problem so I doubt it's a user configuration issue. It's fixed now.

SciresM, 2021-09-17:
Sounds like a user configuration issue.

jltravieso, 2021-09-17:
Perhaps is the way of launching the cfw, have you try injecting the payload (fuse.bin) directly or did you launch hekate and then launch cfw. In some cases the last method can cause issues like the one described.

3DSBricker, 2021-09-17:
Actually you're right. That was what's changed, only it's not the way you would've expected: it gave the error using the official Fusee.bin, but not when using Hekate fss0 with Package3. It is working WITH Hekate.

SciresM, 2021-09-17:
Works using fusee for me.

so...user configuration :P