Atmosphere-NX/Atmosphere #1618: NTP redirect

Ammako, 2021-09-12:
## Feature Request

### What feature are you suggesting?
#### Overview:
Given that `https://aauth-%.ndas.srv.nintendo.net/v1/time` is not accessible when nintendo.net is blocked, via dns or hosts, I suggest the ability to redirect time sync requests to a different ntp server.

#### Smaller Details: 
Perhaps leave `https://aauth-%.ndas.srv.nintendo.net/v1/time` as the default, and just have an optional setting for replacing the time sync server address with a different one.

#### Nature of Request:  
Addition

### What component do you feel this would best fit within?
Not sure, you tell me ;p whatever might be taking care of nim/patching it?

### Why would this feature be useful?
Because hosts/dns should be the preferred way to block access to Nintendo's servers, not incognito, but by blocking Nintendo servers you lose the ability to automatically set network time. There are several homebrew apps than can query unofficial servers to reset NetworkSystemClock, but why when this is built-in to HOS?

Unblocking `aauth-%.ndas.srv.nintendo.net` is not an option, because you can't guarantee that ntp is or will ever be the only thing going through that domain. One could (in theory?) run their own ntp server at `127.0.0.1/v1/time` on a server at home and redirect `aauth-%.ndas.srv.nintendo.net` to the local server, but that's a bit overkill.

SciresM, 2021-09-12:
`One could (in theory?) run their own ntp server at 127.0.0.1/v1/time on a server at home and redirect aauth-%.ndas.srv.nintendo.net to the local server, but that's a bit overkill.`

Yes, this is what one should do.

SciresM, 2021-09-12:
More to the point, I don't really want to develop and bundle a whole lot of nim patches/figure out a sane way to do this and implement it when there's a thing you can already do using dns.mitm that will work.

I would take a pull request a little more seriously, but if you want to pull request something like this please reach out to me (SciresM#0524) on discord once you've got a design ready and we can can talk shop about what shape an implementation of your specified design would take.

closed 2021-09-12 by SciresM

Ammako, 2021-09-12:
For the time being, personally I'd probably just stick with incognito. I don't really have the time to dig into that right now, especially given that I'd have to learn how all the AMS stuff works from scratch. Maybe one day, but chances are someone with the know-how and the interest will get there first.