Atmosphere-NX/Atmosphere #1619: blank_prodinfo_emummc Does Not Blank Serial Number

admiralspeedy, 2021-09-12:
## Bug Report

### What's the issue you encountered?

Setting blank_prodinfo_emummc=1 in exosphere.ini does not blank the console serial number as expected.

### How can the issue be reproduced?

- Create in **SD:\atmosphere\config\exosphere.ini** with the following contents:
```
[exosphere]
blank_prodinfo_emummc=1
```
- Inject fusee.bin and boot Atmosphere.
- Go to Settings > System > Serial Information and the console's serial number will still be displayed.

### Crash Report

NA

### System Firmware Version

12.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: **Chainload fusee via Hekate v5.6.0**

- Official release or unofficial build: **Official release version 1.0.0**

- Do you have additional kips or sysmodules you're loading: **No**
 
- Homebrew software installed: **None**

- EmuMMC or SysNAND: **Partition-based EmuMMC**

### Additional context?

This was discovered on an unpatched Erista unit with a clean install of Atmosphere 1.0.0 and other config files, including **override_config.ini** and **system_settings.ini** work without issue.

I have never tried this setting on any previous build of Atmosphere or older firmware version (I didn't even know it existed until tonight) and I also tried adding **blank_prodinfo_sysmmc=1** to the config to see if that had something to do with it, but it didn't.

Interestingly I also tried booting Atmosphere via FSS0 with Hekate's **cal0blank=1** boot entry setting, which overrides both the **blank_prodinfo_sysmmc** and **blank_prodinfo_emummc** exosphere config items and this results in the serial number being blanked with zeroes correctly, so something tells me that fusee is causing the exosphere.ini to be ignored.

SciresM, 2021-09-12:
exosphere.ini goes on the SD card root, not inside atmosphere/config.

Hekate will also use settings in it if present in the right place.

SciresM, 2021-09-12:
(This file is unique in where it goes, because people often delete the atmosphere folder when updating, and I wanted to minimize the risk of users losing blanking config by mistake when updating atmosphere)

SciresM, 2021-09-12:
Good report btw. I appreciate your using the issue templates fully and correctly.

closed 2021-09-12 by SciresM

admiralspeedy, 2021-09-12:
Ha, I guess I totally missed the fact that it goes on the SD root, I just re-read the docs and saw it right before you closed this. I was going to ask why it would be on the root but you also answered that.

I made sure to use the template because I'm a software developer myself and I see the ridiculous reports you get on a regular basis 😂