Atmosphere-NX/Atmosphere #1620: Daybreak should ignore filenames beggining with .

newkind, 2021-09-12:
## Bug Report

### What's the issue you encountered?

When copying the firmware files to SD card using Finder in MacOS, system automatically creates extra hidden files with exactly the same filenames but they do begin with `.`

This confuses Daybreak and it returns an error message: 
`Failed to get update information, result: 0x00234a02`

Using terminal and deleting all files beggining with `.` inside of that folder, solves the issue and allows daybreak to function properly.

### How can the issue be reproduced?

1. Use MacOS
2. Download firmware and extract it
3. Copy it to a folder on an SD card
4. Try to use Daybreak and point it to that directory 

### Crash Report

N/A

### System Firmware Version

HOS 12.1.0

### Environment?

- Atmosphere 1.0.0 + fusèe
- Hekate 5.6.0
- Partition based EmuMMC

### Additional context?

N/A

Masamune3210, 2021-09-12:
Please check for similar issues before opening a new one, this has already been covered before

Masamune3210, 2021-09-12:
https://github.com/Atmosphere-NX/Atmosphere/issues/1487

closed 2021-09-12 by SciresM