Atmosphere-NX/Atmosphere #1621: Panic ocurred while running Atmosphere

ivoHellSinky, 2021-09-15:
### What's the issue you encountered?

I updated Atmosphere from vesion 19.0.5 to latest 1.0.0 from the AiO Switch Updater. After that, it restarted correctly into Hekate. But when I booted either sysNand or emuNand, it would display the Atmosphere logo and promptly display the following error message:
Panic ocurred while running Atmosphere.
Title ID: 0100000000000008
Error: std::abort(0xFFE)
Press POWER to continue.</br>

Apart from updating Atmosphere and opting out of changing muy config files from the AiO updater, I didn't change anything.</br>

### How can the issue be reproduced?

1. Start Atmosphere from sysNand in ver. 19.0.5
2. Boot to homebrew pressing R while opening an app
3. Open AiO Switch Updater
4. Press "Update Atmosphere"
5. Deny the prompt asking to reset config files
6. Accept the prompt asking for compatibility files
7. Accept the prompt asking to update Hekate
8. Deny the prompt asking to change the Hekate config files
9. Restart the Switch into Hekate
10. Boot either emuNand or sysNand</br>

### Crash Report

[report_0755e67e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7168803/report_0755e67e.zip)
(It was a .bin file and there was no .txt linked to the report...)

### System Firmware Version

12.1.0</br>

lorek123, 2021-09-15:
Most likely Hekate is trying to boot the fusee secondary not the package3. Check your Hekate config . And this is not the atmosphere issue in general

ivoHellSinky, 2021-09-15:
> Most likely Hekate is trying to boot the fusee secondary not the package3. Check your Hekate config . And this is not the atmosphere issue in general

yeah i just found out and fixed it like 5 seconds ago, what a duoy I am. Thanks anyways!

closed 2021-09-15 by ivoHellSinky