Atmosphere-NX/Atmosphere #1622: Not compatible with update 13.0.0

Santiav90, 2021-09-15:
## Bug Report

### What's the issue you encountered?

I'm still not sure if this is where I'm supposed to post this, but I do it for most updates and people search for a post like this to see if an update has been released already, but yeah, atmosphere crashes with the latest switch update.

### How can the issue be reproduced?

update your switch

### System Firmware Version

13.0.0
### Environment?

Running the latest atmosphere update

sapycola, 2021-09-15:
Confirm for me... 
Wait next Atmosphere version!

Masamune3210, 2021-09-15:
Its well known and well announced by almost every hacking news or group setting that atmos will almost never work with a new major update. this happens literally every time

Conan179, 2021-09-15:
https://twitter.com/SciresM/status/1437945978657468417

closed 2021-09-15 by SciresM

jajie95, 2021-09-18:
Personaly, I'm a little new on this so I updated my switch to 13.0.0 and after that the appearing error says: A fatal error running Fusee,  unable to identify package1! 
Waiting for update xD