Atmosphere-NX/Atmosphere #1623: 13.0.0 Support

SciresM, 2021-09-15:
Master issue/check list of stuff to do for 13.0.0 changes.

- [ ] Reverse engineer new kernel changes
- [ ] Reverse engineer new erpt changes (new services "sprof:sp", "sprof:bg")
- [ ] Reverse engineer new ncm changes (two new IPC commands)
- [ ] Reverse engineer potential new boot sysmodule changes
- [ ] Update fusee/exosphere with new keydata/values
- [ ] Update erpt to implement the two new sprofile services
- [ ] Implement new changes into mesosphere
- [ ] Implement new NCM changes
- [ ] Miscellaneous reverse engineering
- [ ] Miscellaneous implementation work/updating
- [ ] Make sure libnx is updated for anything relevant.

nEdAy, 2021-09-16:
If it is a dual system, can you upgrade the NormalSYS first?

SciresM, 2021-09-16:
Locking this issue to prevent spam from people waiting for update.