Atmosphere-NX/Atmosphere #1628: DNS Mitm with emummc id not work

Domyx-01, 2021-09-19:
### What's the issue you encountered?

Hi, I wanted to use Dns Mitm for emummc via id (because I have more than one emunand on the sd and I don't want to use Dns Mitm on all emunands) but it doesn't work.
The emunand I created it with Hekate and it has its own id assigned thanks to the new update of Hekate 5.6.1 which assigns the id, the id of the emunand I created is 0x32574152 so I called the file to put in the Host folder "emummc_32574152.txt" but I also noticed thanks to the Dns Mitm Log that looks for the file "emummc_0000.txt" so it is not working correctly. 
I leave below the log file of Dns Mitm if you want to check.

- In the "config" folder I have "override_config.ini" with:

[hbl_config]
override_key_0=R

- In the "config" folder I have "stratosphere.ini" with:

[stratosphere]
nogc = 0 

### How can the issue be reproduced?

To reproduce the problem you need to create emunand with Hekate and boot via FSS0:

fss0 = atmosphere / package3
emummcforce = 1
emupath = emuMMC / RAW2

then create the Host file to put in the appropriate folder with your id generated by Hekate and you will notice that Dns Mitm looks for this file "emummc_0000.txt" that I don't have at all on the SD.

### Crash Report

No crash reports, no custom firmware crashes, Dns Mitm simply doesn't work ...

### System Firmware Version

12.1.0

### Environment?

- I use Hekate 5.6.1 (build ebefd1c) to be able to launch Atmosphere 1.1.0 (build 006f802) via FSS0
- Both official builds downloaded from Releases on Github
- Partition Emummc created with Hekate -> Tools -> "Partition SD Card"

SciresM, 2021-09-19:
Sounds like hekate isn't setting the emummc_id field when booting exosphere.

The code definitely does the right thing -- what result do you see in logs when booting with fusee instead of hekate?

Domyx-01, 2021-09-19:
Sorry I don't understand how to send my file here on Github, I try to paste the logs of Dns mitm:

DNS Mitm:
Adding defaults to redirection list.
Selecting hosts file ...
Skipping /hosts/emummc_0000.txt because it does not exist ...
Skipping /hosts/emummc.txt because it does not exist ...
Selected /hosts/default.txt
Redirections:
     `receive-lp1.er.srv.nintendo.net` -> 127.0.0.1
     `receive-lp1.dg.srv.nintendo.net` -> 127.0.0.1

SciresM, 2021-09-19:
I asked what happens when you boot via fusee instead of hekate. Please test that.

Domyx-01, 2021-09-19:
Oh I never use fusee.bin, I use two emunands on my SD ... I'll try ...

Domyx-01, 2021-09-19:
Ok SciresM I tried to boot via fusee.bin and you are right, it works!
It recognized the emummc id and Dns Mitm worked as expected by loading the emummc file.
Then it is definitely a problem related to Hekate, do you know what I should tell CTCaer to solve the problem I open the problem there?

I leave you the Logs below:

DNS Mitm:
Adding defaults to redirection list.
Selecting hosts file ...
Selected /hosts/emummc_32574152.txt
Redirections:
    `* ctest.cdn.nintendo.net` -> 95.216.149.205
    `* conntest.nintendowifi.net` -> 95.216.149.205
    `* nintendoswitch.cn` -> 127.0.0.1
    `* nintendoswitch.com.cn` -> 127.0.0.1
    `* nintendoswitch.com` -> 127.0.0.1
    `* nintendo.ch` -> 127.0.0.1
    `* nintendo.se` -> 127.0.0.1
    `* nintendo.co.za` -> 127.0.0.1
    `* nintendo.ru` -> 127.0.0.1
    `* nintendo.pt` -> 127.0.0.1
    `* nintendo.no` -> 127.0.0.1
    `* nintendo.nl` -> 127.0.0.1
    `* nintendo.it` -> 127.0.0.1
    `* nintendo.hu` -> 127.0.0.1
    `* nintendo.gr` -> 127.0.0.1
    `* nintendo.fr` -> 127.0.0.1
    `* nintendo.fi` -> 127.0.0.1
    `* nintendo.de` -> 127.0.0.1
    `* nintendo.dk` -> 127.0.0.1
    `* nintendods.cz` -> 127.0.0.1
    `* nintendo.be` -> 127.0.0.1
    `* nintendo.at` -> 127.0.0.1
    `* nintendo.co.nz` -> 127.0.0.1
    `* nintendo.com.au` -> 127.0.0.1
    `* nintendo.com.hk` -> 127.0.0.1
    `* nintendo.tw` -> 127.0.0.1
    `* nintendo.co.kr` -> 127.0.0.1
    `* nintendo.es` -> 127.0.0.1
    `* nintendowifi.net` -> 127.0.0.1
    `* nintendo-europe.com` -> 127.0.0.1
    `* nintendo.co.uk` -> 127.0.0.1
    `* nintendo.co.jp` -> 127.0.0.1
    `* nintendo.jp` -> 127.0.0.1
    `* nintendo.net` -> 127.0.0.1
    `* nintendo.com` -> 127.0.0.1
    `receive-lp1.er.srv.nintendo.net` -> 127.0.0.1
    `receive-lp1.dg.srv.nintendo.net` -> 127.0.0.1

SciresM, 2021-09-19:
Yeah I would make an issue on the hekate repo.

closed 2021-09-19 by SciresM