Atmosphere-NX/Atmosphere #1629: Black screen on firmware V13 with latest atmosphere and hekate

Abro61209, 2021-09-19:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
Black screen after atmosphere logo

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
Add all the updated files of atmosphere and hekate to the SD card.

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
V13

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
I used Hekate for tegraRCM GUI and Fusee for the bootloader on the console. Using sysNAND.

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?
I used this guide and followed these extra steps because I was coming from below atmosphere 1.0 and still have a black screen.
I have a day one switch console.

https://nh-server.github.io/switch-guide/extras/updating/
- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-09-19:
1. You didn't use the issue templates 
2. You didn't update your config/atmosphere/hekate correctly, or you have incompatible custom system modules.

SciresM, 2021-09-19:
Please note that this issue tracker is for bugs, not for user support.

closed 2021-09-19 by SciresM

Abro61209, 2021-09-19:
> 1. You didn't use the issue templates
> 2. You didn't update your config/atmosphere/hekate correctly, or you have incompatible custom system modules.

Uhh I did update my config/atmosphere/hekate correctly I literally pointed a guide at the bottom. Idk what incompatible sys modules means.

SciresM, 2021-09-19:
Seek support in the Nintendo homebrew discord. You did not follow the relevant guides correctly, or you have custom sysmodules.