Atmosphere-NX/Atmosphere #1631: I accidentally updated to 13.0.0 ( A fatal error occurred when running Atmosphere.)

justJust1, 2021-09-20:
So I accidentally updated my switch to 13.0.0 and I was trying to reboot my switch by reinjecting the payload, but after some digging around they said I would have to wait for a patch update, so now that I downloaded it today, I got the following message:

A Fatal error occurred when running atmosphere.
Program ID: 010041544D530000
Error Desc: std: :abort( ) called  (0xffe)

Report saved to /atmosphere/fatal_errors/report_000000006e1015b.bin
Press power to reboot.

Mewthree3, 2021-09-20:
Upload the report?

justJust1, 2021-09-20:
Im kinda new to this how do I do that

Mewthree3, 2021-09-20:
Delete the folder with the name like 001000100000

justJust1, 2021-09-20:
the .bin folder right?

Mewthree3, 2021-09-20:
Sorry! I thought I was replying to someone else!
Please upload the .bin file in /atmosphere/fatal_errors/report_000000006e1015b.bin!
I'm super sorry!

Mewthree3, 2021-09-20:
Also, are you aware that there's a pre-release build of atmosphere that supports HOS 13.0.0?
https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.1.0

justJust1, 2021-09-20:
oh no your good
 but here it i
[report_0000000006e1015b.txt](https://github.com/Atmosphere-NX/Atmosphere/files/7193244/report_0000000006e1015b.txt)
s

Mewthree3, 2021-09-20:
Now you'll need to wait for someone who works with the Atmosphere code to help you. In future cases please use the issue template. It helps the people trying to help you a lot! I made one for you to fill out, and I filled some stuff that I knew from what you had explained.
For the stuff that I put Unknown on, please copy what I wrote and fill it in!

## Bug Report

### What's the issue you encountered?

So I accidentally updated my switch to 13.0.0 and I was trying to reboot my switch by reinjecting the payload, but after some digging around they said I would have to wait for a patch update, so now that I downloaded it today, I got the following message:

A Fatal error occurred when running atmosphere.
Program ID: 010041544D530000
Error Desc: std: :abort( ) called (0xffe)

Report saved to /atmosphere/fatal_errors/report_000000006e1015b.bin
Press power to reboot.

### How can the issue be reproduced?

Update Atmosphere to 13.0.0, with something messed up by the title 010041544D530000.

### Crash Report

[report_0000000006e1015b.txt](https://github.com/Atmosphere-NX/Atmosphere/files/7193283/report_0000000006e1015b.txt)
### System Firmware Version

13.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Unknown
- Official release or unofficial build: 
  - Atmosphere 12.1.0 (?)
- Do you have additional kips or sysmodules you're loading: Unknown
- Homebrew software installed: Unknown
- EmuMMC or SysNAND: Unknown
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
Unknown

Masamune3210, 2021-09-20:
Could try backing up sd card, completely wiping it and then extracting a fresh copy of the atmosphere files to make sure the copy worked correctly. Alternatively try running the archive bit fixer

justJust1, 2021-09-20:
Ive tried wiping the sd card a couple of times, but what is this second option

Mewthree3, 2021-09-20:
Go to the Hekate menu, and click Tools.
From there go to Tools, then Unset Archive BIt. This fixes corruption stuff and won't hurt, only help.

justJust1, 2021-09-20:
ok ill try that

justJust1, 2021-09-20:
all it says is 0 unset, 0 set! after i do it

Mewthree3, 2021-09-20:
In that case it didn't help.
So you've said that you've tried SD card wipes. Is your card formatted to Fat32 or exFAT?
Also, are you extracting the latest atmosphere every time, or are you using your backed up atmosphere folder?

justJust1, 2021-09-20:
I have my sd card in exFAT

justJust1, 2021-09-20:
and I am extracting from the new update that happened today

Mewthree3, 2021-09-20:
That's probably it!!
exFAT SD cards are a huge bad thing in the Switch homebrew community.
You can use Hekate to format your SD into FAT32. Odds are if you do that you'll be fine! Because exFAT ones are really prone to corruption (on the Switch), please use FAT32!

justJust1, 2021-09-20:
Ok let me try that real quick

justJust1, 2021-09-20:
THANK YOU SOO MUCH IT WORKED I THOUGHT I BRICKED MY SWITCH

Mewthree3, 2021-09-20:
Gyah!
I’m so glad that it worked! Switching to Fat32 is weird like that, for some reason it just works.
You have any other questions? 
Also, it seems like you’re a beginner so to make sure you don’t have anything you missed out on go to switch.hacks.guide and switch.homebrew.guide
Best wishes man!

justJust1, 2021-09-20:
Thx man for all ur help

closed 2021-09-20 by justJust1

Masamune3210, 2021-09-20:
> Gyah! I’m so glad that it worked! Switching to Fat32 is weird like that, for some reason it just works. You have any other questions? Also, it seems like you’re a beginner so to make sure you don’t [...]

For future reference, exfat sucks on the switch because Ninty decided to ignore the spec and made possibly one of the worst exfat drivers ever. Never use exfat on the switch unless absolutely necessary

weirdal3333, 2021-09-20:
Someone smarter than us needs to replace ninty's exfat driver with one that actually works. Maybe we could borrow code from here: https://github.com/arter97/exfat-linux

SciresM, 2021-09-20:
@weirdal3333 if you leave a comment on a closed/inactive issue again like this without some justification I will block you from using the issue tracker.

iMonZ, 2021-10-01:
> Someone smarter than us needs to replace ninty's exfat driver with one that actually works. Maybe we could borrow code from here: https://github.com/arter97/exfat-linux

That's actually not necessary.
The brand-new official exFAT is included in the latest Linux firmware, so Nintendo just need to update their Kernel and that's it.

SciresM, 2021-10-01:
@iMonZ Uh.

1. Nintendo does not use linux
2. Nintendo uses a custom microkernel, filesystem code is in userspace not in the kernel (see mesosphere for a kernel reimplementation)
3. Nintendo uses a proprietary Japanese driver called prFILE2, not open drivers.

Totk07, 2024-04-18:
J'ai un problème lorsque je lance Zelda Tears of The Kingdom avec des astuces, c'est un mod appelé Actor Spawn Utility et lorsque je lance le jeu, j'ai ce problème. Une erreur fatale s'est produite lors de l'exécution d'Atmosphere.
ID de programme 010000000000000d
Description de l'erreur : std : abort( ) appelé (0xffe)

Rapport enregistré dans /atmosphere/fatal_error/report_0000000027b2f065.bin
Appuyez sur POWER pour redémarrer.