Atmosphere-NX/Atmosphere #1632: Error Code: 2162-0002 (0x4a2)

zzpixel, 2021-09-20:
So I updated to the new pre-release and on boot a got an error message for the program "0100000000001000". I'm not sure what caused this but it is to be expected since its a pre-release.

Mewthree3, 2021-09-20:
It’s your custom theme. Delete the folder!

zzpixel, 2021-09-20:
Yeah so I deleted it and now it works. I still have to wait till full release to reinstall my themes. Thanks for the help.

Mewthree3, 2021-09-20:
Yep! I actually was able to re-make my own theme (I use a custom one) and it worked fine on HOS 13, just ig I had to remake it.
[sangatsu_nxthemepackv2.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7193247/sangatsu_nxthemepackv2.zip)
.

CoolAn127Gamer, 2021-09-20:
it works perfect. except that some of the games crashes and NRO apps

Masamune3210, 2021-09-20:
That's beyond the scope of this issue. Redownload games If they are corrupted or not working from the eShop, and make sure your nro's are up to date

hexkyz, 2021-09-20:
Closing down as the issue has been resolved.

closed 2021-09-20 by hexkyz