Atmosphere-NX/Atmosphere #1635: Switch Lite SX Lite

synbios88, 2021-09-20:
### What's the issue you encountered?

[ Updated SwitchLite to latest Atmos, now on my Lite it wont power up, I have to remove SD card to get boot.dat screen then press power again otherwise i get nothing.
[ Did you make any changes related to Atmosphère itself? ] No changes other update from 12.1.0.

### How can the issue be reproduced?

Power off device then try power on..black screen. remove sd card power on to boot.dat, then off again. re insert sd card and power on atmos will boot. 

### Crash Report
NA

### System Firmware Version

12.1.0 OFW and 13.0 CFW

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee.bin
- Latest release of Atmos released today.
- Do you have additional kips or sysmodules you're loading: Na
- Homebrew software installed: NA
- EmuMMC or SysNAND: Emunand
  - [ If using an EmuMMC, include whether it's partition-based or file-based. Partition.

SciresM, 2021-09-20:
Fusee.bin can't be loaded directly, you need to chainload it via hekate when booting on Mariko hardware.

SciresM, 2021-09-20:
This may change in the future ^^

closed 2021-09-20 by SciresM