Atmosphere-NX/Atmosphere #1636: Orange screen freeze

WIIU99, 2021-09-20:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

Orange screen after nintendo logo

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

Happens every time
### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

No crash report

### System Firmware Version

13.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]
  - 
Latest atmosphere, nothing else on sd card
### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
Completely wiped sd card and downloaded atmosphere fresh, switch freezes with a completely Orange screen after Nintendo logo, works booting nornally

masagrator, 2021-09-20:
You didn't fill out issue template properly. 

Beside

> How can the issue be reproduced?
> Happens every time

Are you joking?

SciresM, 2021-09-20:
Orange screen occurs when a hardware device throws an exception. Not really sure there's anything we can do for you.

S1ngyy, 2021-09-20:
Yeah full orange screen usually happens when there is some hardware fault, usually with the WIFI-chip, e.g. dead chip or broken/cracked solder joints. Would definitely be weird if this was caused by Atmosphere.

WIIU99, 2021-09-20:
Why does it not happen on official firnware

SciresM, 2021-09-20:
I don't know. Does it change if you use a different bootloader? (Hekate instead of fusee, or vice versa)

SciresM, 2021-09-20:
You didn't fill out the issue templates properly, please provide more information about your environment.

WIIU99, 2021-09-20:
Don't know what else to say

Firmware 13.0.0
Atmosphere 1.1.0
No emummc
Nothing else on sd card, formatted it before putting atmosphere on

SciresM, 2021-09-20:
Can you try using different bootloader? Which bootloader are you using?

WIIU99, 2021-09-20:
Bootloader

WIIU99, 2021-09-20:
Ok just got a dark blue screen

masagrator, 2021-09-20:
> Bootloader

This doesn't answer the question.
What you have now used? Hekate, fusee, other payload?

WIIU99, 2021-09-20:
Could it be something to do with derrek

SciresM, 2021-09-20:
I'm closing this because a) it seems like a hardware problem, b) it seems like you're trolling.

closed 2021-09-20 by SciresM

WIIU99, 2021-09-20:
I'm not trolling and it doesn't happen in regular switch?

SciresM, 2021-09-20:
If you're not trolling, please directly answer the bootloader questions, thanks.

But also, there's kind of just nothing we can really do if you're getting hardware exceptions thrown.

WIIU99, 2021-09-20:
Im using fusee and its not happening with out atmosphere

WIIU99, 2021-09-20:
Fuck it i give up