Atmosphere-NX/Atmosphere #1637: std::abort() called(0xffe)

zudu, 2021-09-20:
I updated to  pre-release, boot a got a fatal error message for the program 

A fatal error occurred when running Armosphere.
Program ID:  010041544d530000
Error Desc: std::abort() called(0xffe)

not sure what caused this

parnz, 2021-09-20:
try this
https://github.com/Atmosphere-NX/Atmosphere/issues/1631#issuecomment-922597232

SciresM, 2021-09-20:
You didn't use the issue templates, and didn't upload the fatal error report.

closed 2021-09-20 by SciresM

Totk07, 2024-04-18:
I have a problem when I launch Zelda Tears of The Kingdom with cheats is a mod called Actor Spawn Utility and when I launch the game I have this problem. A fatal error ocurred when running Atmosphere.
Program ID 010000000000000d
Error Desc: std: :abort( ) called (0xffe)

Report saved to /atmosphere/fatal_error/report_0000000027b2f065.bin
Press POWER to reboot.