Atmosphere-NX/Atmosphere #1638: Black screen after firmware update

Jrazorman, 2021-09-20:
I installed the update for atmosphere 1.1.0 and all worked fine. I hen proceeded to dump my firmware with "Daybreak" like I have done before. The dump went smooth, going back into the firmware went smooth, using "Daybreak" to read and verify the dump went smooth. After the restart atmosphere only loads to the second splash screen, then goes blank. I can still load HEKATE, but execution on it are hit and miss. My switch still loads normal in system mode and reads all my games fine.  Please help.

masagrator, 2021-09-20:
1. You didn't use issue template
2. Delete custom sysmodules from `atmosphere/contents` and try again. If you don't know what those are, you can just delete whole `contents` folder

Jrazorman, 2021-09-20:
THANK YOU THANK YOU THANK YOU!   I thought really messed up!  Thank you for getting back to me so fast, deleting the contents folder worked like a charm!   and it even seem to boot up faster!   I am very sorry I did not use the issue template, I know for next time.

hexkyz, 2021-09-20:
Closing down as the issue has been resolved.

closed 2021-09-20 by hexkyz