Atmosphere-NX/Atmosphere #1640: "Fatal error: [NXBOOT]: Unable to identify package1!"

jlopez2310, 2021-09-20:
Updated the nitendo switch to the latest version 13.0.0.
And I did the last download of the atmosfere and installed everything correctly with the "RCM Loader" and the screen goes all black and if I try again it looks like "Fatal error: [NXBOOT]: Unable to identify package1!".

What should you do to get the atmosphere back on track?

Mewthree3, 2021-09-20:
Use the issue template next time please!
How’re you booting? Hekate or Fusee?

Aura314, 2021-09-20:
you have not used the details of the template please open a new one because without a template it will simply be closed please folks please do yourself a favor it saves you time and you don’t have to rewrite it over and over again your problem with atmospehre!

jlopez2310, 2021-09-20:
I'm starting with Fusee.
And how do I do this from the model details?

abdelmoumen03, 2021-09-20:
Hi guys if u use new atmosphe downgrade the consol to 12.1 and wait atmosphere for 13.0

Aura314, 2021-09-20:
The new atmosphere for Hos 13.0.0 has long been published!

abdelmoumen03, 2021-09-20:
There is a loot of beugs so still waiting the final version of atmosphere

jlopez2310, 2021-09-20:
At this moment it already appears to me "Fatal error: failed to stat stage2 (fusee-secondary.bin)!"

SciresM, 2021-09-20:
1. You didn't use the issue templates 
2. This is because you didn't update properly.

@abdelmoumen03 there have been zero actual bugs so far. Just user errors.

closed 2021-09-20 by SciresM