Atmosphere-NX/Atmosphere #1642: Nintendo Switch Lite MID-WAY Boot Failure

AMGe350, 2021-09-20:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Switch Lite fitted with SX Lite chip, all functioning & working correctly with 12.1.0 & Atmosphere 1.0.0. Updated to firmware 13.0 & Atmosphere 1.1.0 & the unit will successfully chainload from Hekate & then the blue Atmosphere screen & after three seconds the package 3 screen appears & then NOTHING else! It's just a dark screen!! I have also copied across from another Mariko unit (Which is on 13.0 & Atmosphere 1.1.0 ALL the necessaryfiles but its still not completely booting up!!]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-09-20:
Make sure you actually updated atmosphere completely...my Lite boots without issues.

SciresM, 2021-09-20:
Also, you didn't fill out the issue template fully. Reopen with it filled out.

closed 2021-09-20 by SciresM

AMGe350, 2021-09-20:
> Make sure you actually updated atmosphere completely...my Lite boots without issues.

Thanks for your immediate response, & by the way my apologies for not mentioning earlier, what a true LEGEND you are to the Switch scene. My friend I always completely delete my Atmosphere folder, WHENEVER a new release has been introduced! I'm ripping my hair out at the moment, because my Erista, Mariko, & even my original unpatched Switch, all flowing along beautifully on Atmosphere 1.1.0 & firmware 13... Is there anything else you could suggest Mr LEGEND??

SciresM, 2021-09-20:
Please don't act that way.

Try booting via hekate directly to see if it makes a difference.