Atmosphere-NX/Atmosphere #1644: Black screen after starting games, and doesn't read game card

httpak, 2021-09-21:
## Bug Report

### What's the issue you encountered?

I switch from SXOS to Atmosphere 1.0.0 and it worked perfectly, until I accidentally switch to OFW and updated it to 13.0.0. And back to CFW when I start the game, I press the update software before I realized I shouldn't. The CFW goes online for some reason.

I then downgrade my CFW back to 10.2.0 and upgrade to Atmosphere 1.1.0 and Hekate 5.6.0 and it seems alright, I can load to the game selection screen, until I press any game and none of them worked. Digital games would go black screen after selecting user, and physical games would say the card can't be read.

### How can the issue be reproduced?

Using the latest Atmosphere and Hekate versions, then press any game

### Crash Report

https://gist.github.com/httpak/669f61f13333bb983f368f4b1eab0862

### System Firmware Version

1.0.0 before pressing software update
1.1.0 after the incident now

### Environment?

Launched by hekate
Atmosphere 1.1.0
No additional kips or sysmodules
EmuMMC   partition-based 

### Additional context?
Burnt Fuses : 14 - 0 (HOS: 11.0.0 -12.0.1)

SciresM, 2021-09-21:
You burnt gamecard fuses for 12.x, this means you can never use gamecards on < 12.x ever again.

Upgrade to 12.x/13.x to resume using gamecards, or configure nogc=0 (see atmosphere/hekate docs).

closed 2021-09-21 by SciresM

httpak, 2021-09-21:
Thanks for the comments. I realized the problem after typing the last sentences. Now game card work, but downloaded games still doesn't.

SciresM, 2021-09-21:
Re-download the games from the eShop/check for corrupted data, should fix them.