Atmosphere-NX/Atmosphere #1645: Atmosphère crashing on boot (0xffe)

Ayerbe-Dev, 2021-09-21:
While trying to boot into Atmosphère I got the following:

A fatal error occurred when running Atmosphère.
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_0000000006679f35.bin

Press POWER to reboot

For reference, I mod Smash Ultimate and since my Switch can't currently connect to college wifi, I have to reboot my Switch every time I transfer a file. The only difference between this and the last time I successfully booted was I added 3 lines of code to a plugin I was working on.

Crash Report:

[report_0000000006679f35.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7204440/report_0000000006679f35.zip)

System Firmware Version: 12.1.0
Atmosphère Version: 0.20.1

Atmosphère was launched through hektate.

So far here are the solutions I've already tried:
- Archive Bit Fixer in Hektate menu
- Reinstalling the latest version of Atmosphère that my Switch is compatible with (I assume the versions of Atmosphère that came out after Switch 13.0.0 aren't compatible with 12.1.0, and since I can't connect my Switch to WiFi at the moment, I can't update to 13.0.0)
- Reinstalling the specific file that I had changed before the crash

None of which fixed the issue. Please let me know if I'm missing something.

masagrator, 2021-09-21:
Well, noone will help you with old Atmosphere. Newest one is only supported by issues except if bug exists multiple version to the newest available.

SciresM, 2021-09-21:
1. You didn't really use the issue templates correctly.
2. Support isn't really provided for non-latest atmosphere versions.
3. The report you uploaded is for Atmosphere 0.19.5, not Atmosphere 0.20.1.
4. The report you uploaded is for invalid settings key value store -- you have a corrupt system_settings.ini. Delete or fix it.

closed 2021-09-21 by SciresM