Atmosphere-NX/Atmosphere #1647: Issue updating to latest update

trillyxri, 2021-09-24:
So basically my little brother updated to the 13.0.0 nintendo update like an idiot and I had to wait for the latest atmosphere update went on and downloaded the update replaced all the files in the corresponding folders and booted via pc tegrarcm using the new fusse.bin and I'm getting this
[image]

Glitch-Code-Design, 2021-09-24:
Delete that folder , it's under atmosphere / contents , you or your brother should remove your themes before updating

SciresM, 2021-09-24:
You didn't use the issue templates, and delete your custom theme.

closed 2021-09-24 by SciresM

trillyxri, 2021-09-25:
Appreciate the response gonna fix it today

trillyxri, 2021-09-25:
it wont let me delete the folder lmao i right click nothing pops up @SciresM

trillyxri, 2021-09-25:
nevermind i got it im an idiot