Atmosphere-NX/Atmosphere #1648: Failed to run tsec_keygen firmware

mj9977, 2021-09-25:
## Bug Report

### What's the issue you encountered?

Booting into atmosphere from hekate gives "A fatal error occurred when running Fusee. Failed to run tsec_keygen firmware! Press Power to reboot.".  Pressing power boots into Atmosphere and everything appears to be working normally.
Updated to Atmosphere 1.1.1 and Hekate 5.6.2 
Copied files to sd card from git repositories, overwriting Atmosphere 0.20. 

### How can the issue be reproduced?

Boot into Hekate 5.6.2, select payload, atmosphere, boots to error message.

### Crash Report

N/A

### System Firmware Version

13.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.6.2
- Official release AMS1.1.1
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Goldleaf .9
- EmuMMC partition-based
### Additional context?

- Additional info about your environment:
N/A

SciresM, 2021-09-25:
Don't boot via SX gear, it's not supported.

closed 2021-09-25 by SciresM

mj9977, 2021-09-25:
Cool ok thank you!