Atmosphere-NX/Atmosphere #1651: Randomly Black Screen after Atmosphere Splash scren

antonio-castillo, 2021-09-29:
## Bug Report

### What's the issue you encountered?

Randomly when i inject Fusee to boot Atmosphere after splash screen Atmosphere displays black screen and not continues to Nintendo logo.
I updated Sysnand and Emunand to 13.0.0 throught Daybreak.
I erased all SD and copied again last Atmosphere, sigpatches and update Fusee.

### How can the issue be reproduced?

It is randomly when i try to boot throught fusee.
Before update to 13.0.0 it didn't happen.

### Crash Report

### System Firmware Version

Version 13.0.0 and AMS 1.1.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: oficial
  - [ Official release version x.x.x (or) unofficial build ] 1.1.1
- Do you have additional kips or sysmodules you're loading: NO
- EmuMMC or SysNAND: EmuMMC, partition based

### Additional context?
I use dongle to inject payload.

zemixe, 2021-09-29:
> sigpatches

Piracy is not supported.

antonio-castillo, 2021-09-29:
> > sigpatches
> 
> Piracy is not supported.

I'm sorry is just to give more info.
This happen before to install sigpatches too.

dragon2snow, 2021-09-29:
Remove or upgrade some of the plug-ins you use

antonio-castillo, 2021-09-29:
> Remove or upgrade some of the plug-ins you use

I don't use any plugin

SciresM, 2021-10-04:
This issue really only occurs when custom sysmodules are installed...I suspect you have them but don't realize so.

I recommend seeking support on discord (nintendo homebrew or reswitched), if that fails we can talk about creating a new issue.

closed 2021-10-04 by SciresM