Atmosphere-NX/Atmosphere #1652: RomFS Patches

fruityloops1, 2021-09-29:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Same function as the `/atmosphere/contents/game id/romfs` folder, but it applies patch files to the original ones</br>

#### Smaller Details: 
- As IPS/IPS32 patches are not very good at removing copyrighted content from the patches, another type of patch file format would be preferred. (BPS/xdelta)</br>

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>

### Why would this feature be useful?
- Some files in game mods can't be fully distributed (legally) and would have to be manually patched by the user, this would make it way easier</br>
- Faster mod load times(?) </br>
- Smaller mod download sizes</br>

masagrator, 2021-09-29:
"Faster mod load times(?)"

Definitely not. xdelta patches will definitely slow down loading files because of its design.

SciresM, 2021-09-29:
Closing this because it's technically infeasible.

For what it's worth, I implemented romfs-patches on the SD card in 2019...the performance hit is so massive that games start becoming unpleasant to play.

We are constrained by technical limitations -- it is too slow to apply patches as files are read, we don't have enough memory to keep arbitrary-patch-count in RAM, etc. Happy to hold a conversation about the technical problems on discord.

Btw Atmosphere only uses IPS by design, because IPS is the only "standard" patch format which does not validate the data it's patching, which is highly desirable.

closed 2021-09-29 by SciresM